Bodmer/TFT_eSPI #1504: [v2.4.11] non-trivial designated initializers not supported

cyberman54, 2021-12-22:
```
In file included from .pio/libdeps/dev/TFT_eSPI/TFT_eSPI.cpp:20:0:
.pio/libdeps/dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA(bool)':
.pio/libdeps/dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
   };
   ^
```

Bodmer, 2021-12-22:
It looks like there is something that has not been defined in your setup, so more details are needed.

One or more of the capital letter  named values in the structure below is not defined:
```
spi_bus_config_t buscfg = {
    .mosi_io_num = TFT_MOSI,
    .miso_io_num = TFT_MISO,
    .sclk_io_num = TFT_SCLK,
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
    .max_transfer_sz = TFT_WIDTH * TFT_HEIGHT * 2 + 8, // TFT screen size
    .flags = 0,
    .intr_flags = 0
  };
```

cyberman54, 2021-12-23:
Defines won't fix this.
`TFT_MOSI`, `TFT_MISO`, `TFT_SCLK ` are defined by my application.
`TFT_WIDHT` and `TFT_HEIGHT` are defined in `ILI9341_Defines.h`.

It looks more like a syntax / compiler issue with the nested struct / union?

cyberman54, 2021-12-24:
Turned out, this was an arduino-espressif32 framework issue. Sorry for misplacement here. Closed.

closed 2021-12-24 by cyberman54

thorrak, 2022-02-08:
@cyberman54 How did you end up resolving this? I'm encountering the same problem, when running the following platform/platform package/framework:

```
platform = espressif32
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
framework = arduino
```