Bodmer/TFT_eSPI #1505: Nonesense pin out numbers

themagicmaster, 2021-12-23:
Help please using ESP32 wroom 32 and 7789 display.
Every time I upload the test code it reports back nonesense pin outs this is what I get 
MOSI    = GPIO 2
MISO    = GPIO 5
SCK     = GPIO 15
TFT_DC   = GPIO 5
TFT_RST  = GPIO 2
All defined with the default pin numbers and read out says it is ESP32.
Any help appreciated

Bodmer, 2021-12-24:
It appears the compiler is not using the setup file you have selected.Check User_Setup_Select.h is only calling up ONE setup file. Post a copy of the setup file you are expecting to be used.

themagicmaster, 2021-12-26:
Thanks for replying.
Firstly I changed the computer I was using, tried on my windows 10 machine and now the test reports the correct pinout.  I also chose another pin designation away from pin 2 as believe this is the led control on this board. Great I thought, all looking good. Tried the colour test. Blank display, apart from being backlit.
Also tried another display in case the first one  was faulty, still not working!

Bodmer, 2021-12-27:
Since the original issue is solved I will close. Please open a new issue with a detailed description of your setup and connections.

closed 2021-12-27 by Bodmer