Bodmer/TFT_eSPI #1507: import of ILI9488 examples from Arduino to PlatformIO

gue21, 2021-12-26:
when importing the examples **_Keypad_480x320_** or **_Touch_Controller_Demo_** for an ILI9488 TFT from Arduino IDE (here both works) I get errors in PlatformIO (Visula Studio Code) during compilation
`...
In file included from C:/Users/MiGu/Documents/PlatformIO/Projects/ESP32_Keypad_480x320/src/Keypad_480x320.ino:25:0:
.pio\libdeps\esp32doit-devkit-v1\TFT_eSPI/Extensions\Touch.h:4:2: error: expected unqualified-id before 'public'
  public:
  ^
.pio\libdeps\esp32doit-devkit-v1\TFT_eSPI/Extensions\Touch.h:22:2: error: expected unqualified-id before 'private'  
  private:
  ^
C:/Users/MiGu/Documents/PlatformIO/Projects/ESP32_Keypad_480x320/src/Keypad_480x320.ino: In function 'void loop()': 
C:/Users/MiGu/Documents/PlatformIO/Projects/ESP32_Keypad_480x320/src/Keypad_480x320.ino:122:25: error: 'class TFT_eSPI' has no member named 'getTouch'
   boolean pressed = tft.getTouch(&t_x, &t_y);
...
`
Compiling Touch_Controller_Demo I only get the errors `error: 'class TFT_eSPI' has no member named ...` for the touch functions

Bodmer, 2021-12-27:
"has no member..." means that the TOUCH_CS pin has not been defined.Check your platform.ini file or setup file.

gue21, 2021-12-27:
TOUCH_CS is defined in user_setup.h
The problem is in touch.h. There is nothing before 'public' (no class definition or something else).
I copied public and private section into the public and private section of TFT_eSPI.h and everything is working.
I do not understand why it is working in the Arduino environment

closed 2021-12-27 by gue21

Bodmer, 2021-12-27:
The touch functions are already called up within the class definition section [here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L813-L815). The library has been used successfully by PlatformIO users so it is not clear to me why it is not working for you.

gue21, 2021-12-27:
` #include "Extensions/Touch.h"` was inserted by PlatformIO import also into the main module. and so "private" and "public" statements were in the middle of other declarations!
I removed it - now it works

**thanks for your help** - I did not see this #include in the TFT_eSPI.h

shablam, 2022-10-25:
> ` #include "Extensions/Touch.h"` was inserted by PlatformIO import also into the main module. and so "private" and "public" statements were in the middle of other declarations! I removed it - now it works
> 
> **thanks for your help** - I did not see this #include in the TFT_eSPI.h

Same issue and I am not using arduino framework though. It seems all files get compiled before being added as includes into the TFT_eSPI.h file. It should just import them in then compile as a group i think but its not. Copying the includes directly into the H file is only thing I can see that would solve the issue unless there is a setting somewhere i cant find.