Bodmer/TFT_eSPI #1508: ESP32 S2 SD card problem

cobungra, 2021-12-26:
Hello Bodmer,
First, thanks for your libraries, I have been using them for a range of educational projects with many hardware setups and great results.
My issue is with an ESP32 S2 that's equipped with ST7789 screen and SD card.
Your library updates have brought this to life for JPGs from SPIFFS but the example file SD_Jpg cannot start SD.begin().
I use SD_CS 10 (which works accessing the SD card using an Adafruit app SD.h
#define HAS_SDCARD
#define SD_CS_PIN 10
#define SD_MOSI_PIN 11
#define SD_SCK_PIN 12  
#define SD_MISO_PIN 13

I'd appreciate any assistance.
Regards
[image]

Bodmer, 2021-12-27:
This occurs because the SD and TFT are on different pins so a single SPI hardware port cannot connected to both.

By default most libraries use the FPSI port.

If you add this line to your TFT_eSPI setup file then the library will use a different SPI port:

#define USE_HSPI_PORT

cobungra, 2021-12-28:
Thanks for your quick response, I appreciate your time.
It made sense to me but hasn't helped. (SD.begin failed!)

`#define ST7789_DRIVER      // Full configuration option
#define TFT_WIDTH  135
#define TFT_HEIGHT 240
#define TFT_CS   34  // Chip
#define TFT_DC   37  // Data Command
#define TFT_RST  38    // Set TFT_RST
#define TFT_BL   33  // LED back-light
#define TFT_MOSI 35
#define TFT_SCLK 36

#define LOAD_GLCD   // Font 1.
#define LOAD_FONT2  // Font 2. Small 16 pixel high font,
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font,
#define LOAD_FONT6  // Font 6. Large 48 pixel font,
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font,
#define LOAD_FONT8  // Font 8. Large 75 pixel font
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above,
#define LOAD_GFXFF  // FreeFonts.

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

#define USE_HSPI_PORT
`
The "Read_User_Setup" confirms these settings.. (glad I found that!)

cobungra, 2021-12-30:
I don't know if you have the time to look again at this.. it's seems a pretty specific problem to this board...

I can see these errors with a higher debug level if it's any help.
Cheers

`
 Testing TJpg_Decoder library
[   561][W][sd_diskio.cpp:174] sdCommand(): no token received
[   661][W][sd_diskio.cpp:174] sdCommand(): no token received
[   761][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1161][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1161][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  1161][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1165][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  1172][E][sd_diskio.cpp:795] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1480][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1480][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
SD.begin failed!
`

bhupiister, 2022-01-03:
@Bodmer I am using M5Stack Core2 and it uses this library. I am facing issues with god knows what. 
I am using SD card and TFT_eSPI on same bus. SD card initializes properly and so does the display. 
But sometimes i face issues sometimes while accessing SD card, and i get an error
`[ 88619][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed`

I read your suggestion of flag `USE_HSPI_PORT`  using  `HSPI `but then it does not load the screen.
I have read somewhere that it is because of `ILI9341 `using half duplex mode and causing error to SD card as they are on the same SPI bus.

SD card initializing script
`SD.begin(SS_PIN, SPI, 40000000);`

Please help if you can on this issue, ill really appreciate any kind of help.

Bodmer, 2022-01-06:
I suspect that 40MHz is too high a frequency for an SPI SD card interface. I think 26MHz is the maximum but I tend to use 10MHz.

cobungra, 2022-01-07:
Sorry to say still no progress.
Dropped the speed to 10MHz...
Thanks again for your attention, it's much appreciated, but I feel that in the big picture it's now looking as if the problem is specific to this board (or to my setup).
Regards

bhupiister, 2022-01-07:
@cobungra 
can you show your setup code? Maybe i can help. I have researched a lot in past few days about using TFT and SD card

cobungra, 2022-01-07:
> @cobungra can you show your setup code? Maybe i can help. I have researched a lot in past few days about using TFT and SD card

Certainly, but it's simply Bodmer's example (that works with other hardware).  Example > TJPEG_Decoder > SD_Card
Are you using a similar S2 board with ST7789 display?
` 
The only thing I change is:
#define SD_CS   10
User_Setup.h is a few messages above...

Bodmer, 2022-01-18:
Unfortunately I cannot explain this. Other users have been successful with SD cards. There appears to be multiple libraries available which confuses the problem.

Sorry I can't help furhter.

closed 2022-01-18 by Bodmer

lyusupov, 2022-01-22:
https://github.com/espressif/arduino-esp32/pull/6103