Bodmer/TFT_eSPI #1511: HUAWEI Y220

olegkaras2, 2021-12-28:
https://drive.google.com/file/d/16EkJdOGfc7byZ-DK9mz8t0qhRsK8fvnm/view?usp=sharing
the display from the HUAWEI Y220 phone is controlled by ESP32 , the ILI9488 library, somehow crookedly shows what could be the reason?

Bodmer, 2021-12-29:
It looks like the screen is not being initialised correctly. Maybe the screen is not an ILI9488 type?

Tyr to find more technical information on the screen.

Although it is great to salvage screens from the cost viewpoint,  it is often difficult to find the technical details and ends up being costly in time.

closed 2021-12-29 by Bodmer

olegkaras2, 2021-12-29:
How do I read the display ID over SPI?

olegkaras2, 2021-12-29:
Can a signal analyzer read the display ID by SPI?

> Похоже, что экран инициализируется неправильно. Может быть, экран не относится к типу ILI9488?
> 
> Тир, чтобы найти более подробную техническую информацию на экране.
> 
> Хотя с точки зрения затрат спасение экранов-это здорово, часто бывает трудно найти технические детали, и в конечном итоге это обходится дорого со временем.

Can a signal analyzer read the display ID by SPI?

Rooholla-KhorramBakht, 2022-06-19:
Did you figure out the issue with the display?