Bodmer/TFT_eSPI #1523: viewport?

mandoman2016, 2022-01-02:
I've written a short program for a touch screen display using a MEGA2560R3. Very successful but sadly something happened with Arduino IDE and oddly the original code will no longer "fit" in the MEGAs memory. I don't understand it but it's time for a re-write anyway.
I'm trying to use your excellent TFT_eSPI library with the ESP32 and so far so good :-)
However the project has several pages with touch input and I'm finding that touch sensitive areas from previous pages remain annoyingly active but invisible. Is there a way around this with perhaps ViewPort???  Or any other idea would also be very welcome. I used UTFT on the MEGA project and this never was an issue.  So I just moved right along transporting the old code fat dumb and happy. The issue was a unwelcome surprise. 
I've done my best with the TFT_eSPI.cpp but can't seem to get a handle on how or why one would use ViewPort. Any insight would be appreciated -- perhaps and example/?? :-)
Thanks TOM

Bodmer, 2022-01-02:
Have a look at the example:
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/Viewport_Demo

mandoman2016, 2022-01-03:
Thanks for that Bodmer. Any thoughts on the touch points "Bleed thru" issue??

Bodmer, 2022-01-03:
The whole screen always remains touch sensitive. It is up to the sketch to determine if the touch coordinates occur within a bounded screen area.

closed 2022-01-03 by Bodmer