Bodmer/TFT_eSPI #1524: Installed 2.4.2, getting odd message

MikeyMoMo, 2022-01-03:
call to 'TFT_eSPI::TFT_eSPI(int16_t, int16_t)' uses the default argument for parameter 1, which is not yet defined 

No change in the code.  Went back to 3.70 and working just fine again.

Bodmer, 2022-01-03:
It looks like TFT_WIDTH and TFT_HEIGHT are not defined for some reason.

This may be processor or display driver specific as I am not seeing that message.

Can you tell me which processor you are using and which display driver.

mrjgreg, 2022-01-06:
Similar here:
Example.....
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
In file included from /home/pi/Arduino/libraries/TFT_eSPI/examples/320 x 240/TFT_graphicstest_one_lib/TFT_graphicstest_one_lib.ino:17:
/home/pi/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:391:25: error: 'TFT_WIDTH' was not declared in this scope; did you mean 'TFT_WHITE'?
  391 |   TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);
      |                         ^~~~~~~~~
      |                         TFT_WHITE
/home/pi/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:391:49: error: 'TFT_HEIGHT' was not declared in this scope
  391 |   TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);
      |                                                 ^~~~~~~~~~
exit status 1
Error compiling for board Raspberry Pi Pico.
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Board RPI Pico2040
ILI9341 display

Setup 60  <now 61> modified for user specific GPIO

All worked OK before update ?

Cheers
MrGreg

Bodmer, 2022-01-06:
Which version of the library are you using?

mrjgreg, 2022-01-06:
Hello
It was the latest update installed on 5 Jan 22
ver 2.4.2
On arduino ver 1.8.13 ( debian repo version )
Cheers
MrGreg

Bodmer, 2022-01-06:
Which board package are you using (Earle Philhower's or the Arduino Mbed one)?

mrjgreg, 2022-01-06:
The Earle Philhower's one

Bodmer, 2022-01-06:
Post your complete setup file.

mrjgreg, 2022-01-06:
Just noticed there is an another update to Earle Philhower board. Will try that first and report back...

Cheers
MrGreg

mrjgreg, 2022-01-06:
Same result again with updated board;

Here is the setup:
xxxxxxxxxxxxxxxxxxxxxxxxxxx

//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER
//#define RM68140_DRIVER
//#define ST7796_DRIVER
//#define SSD1963_480_DRIVER
//#define SSD1963_800_DRIVER
//#define SSD1963_800ALT_DRIVER
//#define ILI9225_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
// bi-directional SDA pin and the library will try to read this via the MOSI line.
// To use the SDA line for reading data from the TFT uncomment the following line:

// #define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 display only

// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display

//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

// For ST7789, ST7735 and ILI9163 ONLY, define the pixel width and height in portrait orientation
// #define TFT_WIDTH  80
// #define TFT_WIDTH  128
// #define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
// #define TFT_HEIGHT 160
// #define TFT_HEIGHT 128
// #define TFT_HEIGHT 240 // ST7789 240 x 240
// #define TFT_HEIGHT 320 // ST7789 240 x 320

// For ST7735 ONLY, define the type of display, originally this was based on the
// colour of the tab on the screen protector film but this is not always true, so try
[... 97 lines cut ...]
// For the RP2040 processor define the SPI port channel used, default is 0
// #define TFT_SPI_PORT 1 // Set to 0 if SPI0 pins are used, or 1 if spi1 pins used

// Define the SPI clock frequency, this affects the graphics rendering speed. Too
// fast and the TFT driver will not keep up and display corruption appears.
// With an ILI9341 display 40MHz works OK, 80MHz sometimes fails
// With a ST7735 display more than 27MHz may not work (spurious pixels and lines)
// With an ILI9163 display 27 MHz works OK.

// #define SPI_FREQUENCY   1000000
// #define SPI_FREQUENCY   5000000
// #define SPI_FREQUENCY  10000000
// #define SPI_FREQUENCY  20000000
// #define SPI_FREQUENCY  32000000
 #define SPI_FREQUENCY  70000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

xxxxxxxxxxxxxxxxxxxxxxxxxxxx
Cheers
MrGreg

Bodmer, 2022-01-06:
You will need to define TFT_MISO, TFT_MOSI and TFT_SCLK for the Pico, but that is not the problem.

Have you changed anything in the User_Setup_Select.h file? For some reason [this line is not being executed ](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L152)even though you have defined the display type.

It may be best to load a clean copy of the library.

mrjgreg, 2022-01-06:
OK.....
FYI Re:" You will need to define TFT_MISO, TFT_MOSI and TFT_SCLK for the Pico, but that is not the problem."
Yes, I was aware. I  was trying them <experimenting> commented out as I have had some previous problems with #defining SPI pins being ignored and reverting to hw defaults of 16,17,18,19.
I'll wipe the Lib and start afresh & report back.

Cheers
MrGreg

Bodmer, 2022-01-06:
OK. In a clean copy [comment out this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22) with //

Then remove // at [start of this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L82).

Make no other changes.

Save the file and run Read_User_Setup example, the output should be like this:
```
[code]
TFT_eSPI ver = 2.4.2
Processor    = RP2040
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 3
MISO    = GPIO 0
SCK     = GPIO 2
TFT_CS   = GPIO 20
TFT_DC   = GPIO 18
TFT_RST  = GPIO 19

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 70.00
[/code]

```
Then edit the pins in Setup60_RP2040_ILI9341.h to suit your setup. Run Read_User_Setup again to check the compiler has picked up the new changes correctly.

mrjgreg, 2022-01-06:
Hmmmmmmmmmm!!

Not done the above yet, But...

"Ghost in the machine"....?
Wiped the lib and reinstalled as suggested, ran my original setup 61 but with SPI defined on 16,17,18,19 as per previous setup.
Now works fine !!! 
This , rather my setup is as it was before the update.
V. curious? as the OP who started the issue had the same symptom?
Can only wild guess that it may have been a corrupt update for what ever reason and is now OK?

Do you still want me to run the above amended  diagnostic?

Cheers
MrGreg

Bodmer, 2022-01-06:
If it is working then that is great. Typically this error only happens if the display type is not correctly defined or something got deleted by accident.

Glad it is working.

Bodmer, 2022-01-06:
@MikeyMoMo Can you also follow the advice above and reload TFT_eSPI.

Open a new issue if the problem persists.

closed 2022-01-06 by Bodmer

mrjgreg, 2022-01-06:
Hello I would just like to confirm I had not done anything with user_setup_select other than pointing at my setup.
Lastly, perhaps most importantly:
V BIG Thank you for both the most excellent Lib and the support. Much appreciated !

Cheers
MrGreg