Bodmer/TFT_eSPI #1529: Conver bmp to array

k008, 2022-01-03:
Hi happy new year!
Tell me, I'm trying to convert bmp to array. Using ImageConverter565, but I get wrong colors when flashing.
How to do it right?

Bodmer, 2022-01-03:
This sounds like an "Endianess" problem, where the byte order is different to that expected.

The library has a function that swaps the bytes during rendering which can be used in setup():

tft.setSwapBytes(true);

By default this is set to false.

The other option is to use a different conversion program or one that allows control of the byte order.

Are you using the [online tool](http://www.rinkydinkelectronics.com/t_imageconverter565.php), or the .exe program for conversion?

Bodmer, 2022-01-03:
I have run a test and converted a BMP using the UTFT ImageConverter565.exe tool. It renders fine in[ this sketch](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/TFT_Flash_Bitmap/TFT_Flash_Bitmap.ino#L46) and I see that setSwapBytes is indeed set to true. So that should be the solution.

closed 2022-01-03 by Bodmer

k008, 2022-01-04:
I am using the **ImageConverter565.exe** program. Indeed, I did not specify **tft.setSwapBytes (true);**, thanks