Bodmer/TFT_eSPI #1534: private -> protected

FrankBoesing, 2022-01-05:
Hi,

i'd like to extend your great library. Some needed functions are "private", so this is not possible. Changing that "private:" into "protected:" would allow to use inheritance and to add extensions on top of your lib.

Is there a chance that you change this?

Thank you in advance,   
Frank

FrankBoesing, 2022-01-05:
p.s. and could you add non-inlined versions of begin\_tft\_write() / end\_tft\_write(), please?

Bodmer, 2022-01-06:
Maybe, a few provisos:
1. The extensions must be published online for Arduino users - explain what is proposed.
2. I suspect not all need to be removed from private - which ones?
3. Code is inline for performance so there would need to be a default inline, not sure how this can be changed in a tidy way without another #define.