Bodmer/TFT_eSPI #1535: Setup outside of library

timothyblue, 2022-01-06:
Is there a good way to use this library and set it up in the project code instead of down in the library code?

Using PlatformIO, all of my libraries in a central location but each project that uses the library may have a different configuration. It would be nice to be able to configure this library in my main.cpp instead of in a central library.

Most of the other libraries I use allow you to pass in parameters to an init() or begin() method.

I realize that this library has a lot of options and so that seems daunting, but at least things like what pins to use seems like it would benefit from this sort of option.

Maybe I am missing something, and this is already available somehow?

Bodmer, 2022-01-06:
You can use platform.ini, see here:

https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/PlatformIO/Configuring%20options.txt

closed 2022-01-06 by Bodmer

timothyblue, 2022-01-06:
Worked like a charm!  Thank you for pointing me in the right direction.

This library is a amazing, thank you!