Bodmer/TFT_eSPI #1537: Pacman Example with Issues

hasenradball, 2022-01-06:
HI Bodmer,

I created an Pacman Example with an ILI9341, but it will not run very well.
Do you maybe have an idea, what could be the issue here?

If I comment out the line:
`tft.fillScreen(ILI9341_BLACK);`
in the loop it runs fast...

I copy the code here:

```
/***************************************************
  ESP8266 PacMan Example on the Wemos mini D1
 ****************************************************/

#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();

/*
// for Wemos D1 Mini use CS->16; DC->15
#define TFT_CS 16
#define TFT_DC 15
#define TFT_MOSI 13
#define TFT_CLK 14
#define TFT_RST -1
#define TFT_MISO 12
*/

#define TFT_LED 5

void pacman(){
  int dir {0};
  int m {0};
  tft.fillScreen(ILI9341_BLACK);
  for (int x=0, y = 15; x < 300; ++x) {
    tft.fillScreen(ILI9341_BLACK);
    tft.fillCircle(x, y, y, ILI9341_YELLOW);
    tft.fillCircle(x+1, 7, 2, ILI9341_BLUE);
    tft.fillTriangle(x, 15, x+15, 15-m, x+15, 15+m, ILI9341_BLACK);
    if(m!= 0) {
      tft.fillTriangle(x, 15, x+15, 15-m, x+15, 15+m, ILI9341_BLACK);
    }
    if(dir == 0) {
      m++;
      if(m == 15) {
        dir = 1;
      }
    }
    if(dir == 1) {
      m--;
      if(m == 0) {
        dir = 0;
      }
    }
    delay(4);
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println("\nILI9341 Pacman-Test!"); 
  
  pinMode(TFT_LED, OUTPUT);
  digitalWrite(TFT_LED, LOW);
  
  tft.begin();
}

void loop(void) {

  tft.setRotation(3);
  pacman();
}
```

hasenradball, 2022-01-10:
Hi Bodmer,

Thanks it works great with the Sprite.
One question about the Library.

Is it also possible to rotate the sprite?

I am thinking about to to the following:

1.) drawing an logo in a sprite

2.) push it 

3.) rotate it for example like a second arrow in a clock.

Is is also possible to draw a section of a circle by center, radius, start, end for example?

Bodmer, 2022-01-10:
Yes, you need to look at the example Rotated_Sprite_1+2.

You would have to write your own routine for sectors of a circle. Have a look at this sketch:

https://github.com/Bodmer/TFT_ILI9341/blob/master/examples/TFT_Pie_Chart/TFT_Pie_Chart.ino