Bodmer/TFT_eSPI #1538: pushRotated not respecting tft's pivot

idorurez, 2022-01-10:
Hi All,

I've been looking over all of the examples, and tested pivot locations by drawing little "X"s on both the tft and sprite and double checked by drawing lines and making sure the text I'm writing to the sprite is showing as expected, but I keep hitting this issue no matter where I set the tft pivot:

pushRotated always pivots on the tft at (0,0), the upper left corner. I can pushSprite without a problem to the correct location, but pushRotated doesn't seem to obey the tft.setPivot(x,y) location.

Anyone experience a similar issue? I must be missing something.

```

void displayIndoorConditions(TFT_eSPI tft, sensors_event_t temp_event, sensors_event_t pressure_event, sensors_event_t humidity_event) {
    tft.fillRect(0, 0, TFT_W/2, (TFT_H/2) - 20, TFT_BLACK);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    
    tft.setCursor(0,0);
    tft.loadFont(AA_FONT_LARGE);
    tft.print((temp_event.temperature * 9/5) + 32, 1);
    tft.println("°");
    
    tft.loadFont(AA_FONT_MED);
    tft.setCursor(0,80);
    tft.print(humidity_event.relative_humidity, 1);
    tft.println("%");

    tft.setPivot(235, 0);     // pivot set on edge of other text
    drawX(235, 0);  // make sure i am aware of where the pivot is

      // Create the Sprite
    spr.setColorDepth(1);       // just black and white mam
    spr.createSprite(160, 100);  // 160 x 100 sprite
    spr.setPivot(0, 100);      // bottom left corner of sprite as pivot
    spr.fillSprite(TFT_BLACK); // Fill the Sprite with black

    spr.loadFont(AA_FONT_SMALL);
    spr.drawString(String(pressure_event.pressure) + " hPA", 0, 0); // test font

    spr.drawLine(0, 0, 160, 100, TFT_WHITE);
    drawXspr(0, 100);
    
    spr.pushRotated(90);
}
```

Bodmer, 2022-01-10:
I patched together this example based on your code and it works OK. Can you also check it works for you.
```
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();           // TFT object
TFT_eSprite spr = TFT_eSprite(&tft); // Sprite object

void setup()   {
  tft.begin();  // initialize
  tft.setRotation(1);
  tft.fillScreen(TFT_WHITE);
}
void loop() {
  int xw = tft.width() / 2; // xw, yh is middle of screen
  int yh = tft.height() / 2;

  tft.fillRect(0, 0, xw, yh - 20, TFT_YELLOW);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);

  tft.setCursor(0, 0);

  tft.setTextFont(2);
  tft.print((50 * 9 / 5) + 32, 1);
  tft.println("`");

  tft.setCursor(0, 80);
  tft.print(57, 1);
  tft.println("%");

  tft.setPivot(235, 0);     // pivot set on edge of other text
  drawX(235, 0);  // make sure i am aware of where the pivot is

  // Create the Sprite
  spr.setColorDepth(1);       // just black and white mam
  spr.createSprite(160, 100);  // 160 x 100 sprite
  spr.setPivot(0, 100);      // bottom left corner of sprite as pivot
  spr.fillSprite(TFT_BLACK); // Fill the Sprite with black

  spr.drawString(String(1000) + " hPA", 0, 0); // test font

  spr.drawLine(0, 0, 160, 100, TFT_WHITE);
  drawXspr(0, 100);

  spr.pushRotated(90);

  while (1);
}

void drawX(int x, int y)
{
  tft.drawLine(x - 5, y - 5, x + 5, y + 5, TFT_RED);
  tft.drawLine(x - 5, y + 5, x + 5, y - 5, TFT_RED);
}
void drawXspr(int x, int y)
{
  spr.drawLine(x - 5, y - 5, x + 5, y + 5, TFT_WHITE);
  spr.drawLine(x - 5, y + 5, x + 5, y - 5, TFT_WHITE);
}

```

This is what the screen looks like:
[image]

idorurez, 2022-01-10:
This worked. I looked a the main difference and it was because I was passing tft around incorrectly. To begin with, that's necessary. After making sure the scope to tft was global, it worked.

Thank you!!!

closed 2022-01-10 by idorurez