Bodmer/TFT_eSPI #1551: Question ILI9488 + RP2040

venice1200, 2022-01-13:
Hi, 
Happy new Year and alle the best for you.

Waveshare offers an nice ILI9488 based TFT for the RP2040 and it looks to me that it uses an SPI->16Bit design like your library supports for the 9486.
`The ILI9486 RPi display must be of the Waveshare design and use a 16 bit serial interface based on the 74HC04, 74HC4040 and 2 x 74HC4094 logic chips.
`
https://www.waveshare.com/wiki/Pico-ResTouch-LCD-3.5
https://www.waveshare.com/w/upload/8/85/Pico-ResTouch-LCD-3.5_Sch.pdf

I just want to ask if your library is able to drive this TFT with the RP2040?

Many thanks for your Library
Regards

Bodmer, 2022-01-13:
Yes, you can drive the display with SPI, I have only tested the ILI9486 though. You may find you need to limit the SPI clock rate to 20MHz, although 32MHz may work.

That interface design requires 16 bit commands so you need to put #define RPI_DISPLAY_TYPE in the setup file.

To use the display you need to defined the display driver and correct pins in the setup file, e.g. #define ILI9488_DRIVER

The display appears to interface with SPI port 1 pins so you also need to include #define TFT_SPI_PORT 1 as the Pico defaults to using SP port 0.

closed 2022-01-13 by Bodmer

venice1200, 2022-01-14:
Hi,
**#define RPI_DISPLAY_TYPE** was the key to get it working.
32MHz is wowking as well.
I haven't tested touch is untested.
The option `#define RP2040_PIO_SPI` doesn't work for me. 
Many thanks.

Here is my working (20MHz) config used with the "Master" from 2022-01-13.

```
// Setup000_RP2040_ILI9488.h
// Works with Waveshare "Pico-ResTouch-LCD-3.5" HAT

//#define RP2040_PIO_SPI
#define RPI_DISPLAY_TYPE

#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

#define TFT_MISO  12  //0
#define TFT_MOSI  11  //3
#define TFT_SCLK  10  //2
#define TFT_CS     9  //20  // Chip select control pin
#define TFT_DC     8  //8   // Data Command control pin
#define TFT_RST   15  //19  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL    13  //       LED back-light

//#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define TFT_SPI_PORT 1 // Set to 0 if SPI0 pins are used, or 1 if spi1 pins used

// #define SPI_FREQUENCY   1000000
// #define SPI_FREQUENCY   5000000
// #define SPI_FREQUENCY  10000000
#define SPI_FREQUENCY  20000000
// #define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  32000000
// #define SPI_FREQUENCY  40000000
//#define SPI_FREQUENCY  70000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000
```

Bodmer, 2022-01-14:
Good to hear it is working for you.

At the moment the PIO option does not support 16 bit commands, this will be added at some point.