Bodmer/TFT_eSPI #1560: TTGO v2.4.2 Working, 2.2.20 Not

anthonyjclarke, 2022-01-16:
Using a TTGO T Display board, all working with the files downloaded from [xxx](https://github.com/Xinyuan-LilyGO/TTGO-T-Display) that has an eSPI fork of v 2.2.20

Using "Setup25_TTGO_T_Display.h"

When I use the latest eSPI cannot get the display to work.

Was wondering if anyone has seen/solved this or what the changes between the versions may have changed.

Thanks

When I run the "Read_User_Setup" from both versions this is the output

TTGO Drivers

17:42:22.896 -> [code]
17:42:22.896 -> TFT_eSPI ver = 2.2.20
17:42:22.896 -> Processor    = ESP32
17:42:22.896 -> Frequency    = 240MHz
17:42:22.896 -> Transactions = Yes
17:42:22.896 -> Interface    = SPI
17:42:22.896 -> Display driver = 7789
17:42:22.896 -> Display width  = 135
17:42:22.896 -> Display height = 240
17:42:22.896 -> 
17:42:22.896 -> R0 x offset = 52
17:42:22.896 -> R0 y offset = 40
17:42:22.896 -> MOSI    = GPIO 19
17:42:22.896 -> MISO    = GPIO 19
17:42:22.896 -> SCK     = GPIO 18
17:42:22.896 -> TFT_CS   = GPIO 5
17:42:22.896 -> TFT_DC   = GPIO 16
17:42:22.896 -> TFT_RST  = GPIO 23
17:42:22.896 -> 
17:42:22.896 -> TFT_BL           = GPIO 4
17:42:22.896 -> TFT_BACKLIGHT_ON = HIGH
17:42:22.943 -> 
17:42:22.943 -> Font GLCD   loaded
17:42:22.943 -> Font 2      loaded
17:42:22.943 -> Font 4      loaded
17:42:22.943 -> Font 6      loaded
17:42:22.943 -> Font 7      loaded
17:42:22.943 -> Font 8      loaded
17:42:22.943 -> Smooth font enabled
17:42:22.943 -> 
17:42:22.943 -> Display SPI frequency = 40.00
17:42:22.943 -> [/code]

Latest eSPI

17:47:52.209 -> 
17:47:52.209 -> [code]
17:47:52.209 -> TFT_eSPI ver = 2.4.2
17:47:52.209 -> Processor    = ESP32
17:47:52.209 -> Frequency    = 240MHz
17:47:52.209 -> Transactions = Yes
17:47:52.209 -> Interface    = SPI
17:47:52.209 -> Display driver = 7789
17:47:52.209 -> Display width  = 135
17:47:52.209 -> Display height = 240
17:47:52.209 -> 
17:47:52.209 -> R0 x offset = 52
17:47:52.209 -> R0 y offset = 40
17:47:52.209 -> MOSI    = GPIO 19
17:47:52.209 -> SCK     = GPIO 18
17:47:52.209 -> TFT_CS   = GPIO 5
17:47:52.209 -> TFT_DC   = GPIO 16
17:47:52.209 -> TFT_RST  = GPIO 23
17:47:52.209 -> 
17:47:52.209 -> TFT_BL           = GPIO 4
17:47:52.209 -> TFT_BACKLIGHT_ON = HIGH
17:47:52.209 -> 
17:47:52.209 -> Font GLCD   loaded
17:47:52.255 -> Font 2      loaded
17:47:52.255 -> Font 4      loaded
17:47:52.255 -> Font 6      loaded
17:47:52.255 -> Font 7      loaded
17:47:52.255 -> Font 8      loaded
17:47:52.255 -> Smooth font enabled
17:47:52.255 -> 
17:47:52.255 -> Display SPI frequency = 40.00
17:47:52.255 -> [/code]

Bodmer, 2022-01-16:
Which version of the Arduino ESP32 board package are you using?

anthonyjclarke, 2022-01-16:
esp32 1.04 and esp8266 2.7.4

anthonyjclarke, 2022-01-16:
Did an upgrade to 1.06 and now working, out of curiosity what changed to make it work? Would you know?

Bodmer, 2022-01-17:
Support for the new ESP32 S2 was added and this has led to issues with older board packages.