Bodmer/TFT_eSPI #1575: support for framework-arduinoespressif32 2.0.2

bongoo1, 2022-01-25:
i tried to go from framework-arduinoespressif32 1.0.6 to 2.0.2 because of issues i have in other parts of my design (and which should be fixed in 2.0.2), but i was not able to build with 2.0.2 because of some issues, the compiler reported for the TFT_eSPI library.
i updated the TFT_eSPI library to the latest version, but it still did not work.
is it correct that the TFT_eSPI library only supports 1.0.6?
and if yes, are there any plans to update for support of 2.0.2?
thanx!

Bodmer, 2022-01-25:
I am using 2.0.2 and do not experirnce any problems. The latest 2.4.2 TFT_eSPI requires 2.0.0 or later.  See also this discussion:
https://github.com/Bodmer/TFT_eSPI/issues/1574#event-5948459838

bongoo1, 2022-01-25:
i'm not so experienced with the arduino environment, but this looks to me like running TFT_eSPI 2.4.25 with arduino 1.0.6:

PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TFT_eSPI> 2.4.25
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <NTP> 1.4.0
|-- <ESP32Ping> 1.7.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <SPI> 1.0
|-- <Preferences> 1.0
Building in release mode

the above image works fine.

but as soon as i add

platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
	
to the platformio.ini file, the result looks like that:

PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.6a7bcab
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TFT_eSPI> 2.4.25
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <NTP> 1.4.0
|-- <ESP32Ping> 1.7.0
|   |-- <WiFi> 2.0.0
|-- <Ethernet> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <SPI> 2.0.0
|-- <WiFi> 2.0.0
|-- <Preferences> 2.0.0
Building in release mode
...
Compiling .pio\build\esp32dev\lib6e6\WiFi\WiFiScan.cpp.o
In file included from .pio/libdeps/esp32dev/TFT_eSPI/TFT_eSPI.cpp:20:0:
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA(bool)':
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
   };
   ^
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:791:3: sorry, unimplemented: non-trivial designated initializers not supported
*** [.pio\build\esp32dev\liba30\TFT_eSPI\TFT_eSPI.cpp.o] Error 1

which looks to me like compiling TFT_eSPI 2.4.25 with arduino 2.0.0, and having a compatibility issue of the TFT_eSPI library with the new arduino framework.

what am i doing wrong?

Bodmer, 2022-01-25:
Some functions have defaults, it seems that the compiler is having trouble with a prototype like this:
```
bool     initDMA(bool ctrl_cs = false);
```
This impiles that "false" is not defined for some reason? You could test this by changing to:
```
bool     initDMA(bool ctrl_cs = 0);
```
But you will then end up with errors elsewhere.

I do not use platformIO so I am not sure where the problem is. Maybe the PlatformIO forum is the best place to ask?

bongoo1, 2022-01-26:
looks like adding the line
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
to platformio.ini solves the issue (but i do not really know what this line does).

closed 2022-01-26 by Bodmer

Bodmer, 2022-01-29:
FYI, see #1588 for resolution.