Bodmer/TFT_eSPI #1577: ESP32 vroom with 1.44 128x128 RED_PCB

podaen, 2022-01-25:
I have tested setup 7 and 47 and keep getting a with screen. I m not using the default wiring but I know it works by only using these librarys and pins

```
//#include <TFT_ILI9163C.h>

#include <Adafruit_ST77xx.h>
#include <Adafruit_ST7789.h>
#include <Adafruit_ST7735.h>

#define TFT_CS        17
#define TFT_RST       19
#define TFT_DC        5
```

I rewired it with your default pins an than it was working but very slow, and it stopt working after a few second. I want to use the wiring like this. How should I config it?

This looks right to me, but clearly there is someting wrong.

```
// Config for two ST7735 128 x 128 displays for Animated_Eyes example

#define ST7735_DRIVER     // Configure all registers

#define TFT_WIDTH  128
#define TFT_HEIGHT 128

// #define ST7735_INITB
// #define ST7735_GREENTAB
// #define ST7735_GREENTAB2
 //#define ST7735_GREENTAB3
// #define ST7735_GREENTAB128    // For 128 x 128 display
// #define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
 #define ST7735_REDTAB
//#define ST7735_BLACKTAB
// #define ST7735_REDTAB160x80   // For 160 x 80 display with 24 pixel offset

//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

//#define TFT_INVERSION_ON
//#define TFT_INVERSION_OFF

//// Generic ESP32 setup
//#define TFT_MISO 19
//#define TFT_MOSI 23
//#define TFT_SCLK 18
////#define TFT_CS    21 // Not defined here, chip select is managed by sketch
//#define TFT_DC    2
//#define TFT_RST   4  // Connect reset to ensure display initialises

#define TFT_MOSI 23//15 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 18//14
#define TFT_CS   17//5  // Chip select control pin  Not defined here, chip select is managed by sketch
#define TFT_DC   5//27  // Data Command control pin
#define TFT_RST  19//33  // Reset pin (could connect to Arduino RESET pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
//#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS
```

Bodmer, 2022-01-26:
Pin 19 should not be used as reset, that will be configured as SPI input MISO.  Allocate and wire reset to another GPIO pin.
Suggested connections:
```
// Generic ESP32 setup
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    21
#define TFT_DC    2
#define TFT_RST   4  // Connect reset to ensure display initialises
```

podaen, 2022-01-26:
I will change them back like this. I think this is corresponding with what you sad... 

> // LED		pin 3v
> // SCK		SCK 18
> // SDA		MOSI 19
> // A0		DC 2
> // RESET	        4
> // CS		21
> // GND		GND
> // VCC		pin 5v

podaen, 2022-01-26:
I have still a white screen.

Bodmer, 2022-01-26:
Run the sketch Read_User_Setup and check the serial monitor output is as expected. Post the output.

podaen, 2022-01-26:
> [code]
> TFT_eSPI ver = 2.4.25
> Processor    = ESP32
> Frequency    = 240MHz
> Transactions = Yes
> Interface    = SPI
> Display driver = 7735
> Display width  = 128
> Display height = 128
> 
> R0 x offset = 2
> R0 y offset = 3
> MOSI    = GPIO 23
> MISO    = GPIO 19
> SCK     = GPIO 18
> TFT_CS   = GPIO 21
> TFT_DC   = GPIO 2
> TFT_RST  = GPIO 4
> 
> Font GLCD   loaded
> Font 2      loaded
> Font 4      loaded
> Font 6      loaded
> Font 7      loaded
> Font 8      loaded
> Smooth font enabled
> 
> Display SPI frequency = 27.00
> [/code]

podaen, 2022-01-26:
Do I need to chang anything in here the TFT_eSPI_ESP32.h?

```
#else
/* Old macros
  // ESP32 low level SPI writes for 8, 16 and 32 bit values
  // to avoid the function call overhead
  #define TFT_WRITE_BITS(D, B) \
  WRITE_PERI_REG(SPI_MOSI_DLEN_REG(SPI_PORT), B-1); \
  WRITE_PERI_REG(SPI_W0_REG(SPI_PORT), D); \
  SET_PERI_REG_MASK(SPI_CMD_REG(SPI_PORT), SPI_USR); \
  while (READ_PERI_REG(SPI_CMD_REG(SPI_PORT))&SPI_USR);

  // Write 8 bits
  #define tft_Write_8(C) TFT_WRITE_BITS(C, 8)

  // Write 16 bits with corrected endianess for 16 bit colours
  #define tft_Write_16(C) TFT_WRITE_BITS((C)<<8 | (C)>>8, 16)

  // Write 16 bits
  #define tft_Write_16S(C) TFT_WRITE_BITS(C, 16)

  // Write 32 bits
  #define tft_Write_32(C) TFT_WRITE_BITS(C, 32)

  // Write two address coordinates
  #define tft_Write_32C(C,D) TFT_WRITE_BITS((uint16_t)((D)<<8 | (D)>>8)<<16 | (uint16_t)((C)<<8 | (C)>>8), 32)

  // Write same value twice
  #define tft_Write_32D(C) TFT_WRITE_BITS((uint16_t)((C)<<8 | (C)>>8)<<16 | (uint16_t)((C)<<8 | (C)>>8), 32)
//*/
//* Replacement slimmer macros
  #define TFT_WRITE_BITS(D, B) *_spi_mosi_dlen = B-1;    \
                               *_spi_w = D;             \
                               *_spi_cmd = SPI_USR;      \
                        while (*_spi_cmd & SPI_USR);

  // Write 8 bits
  #define tft_Write_8(C) TFT_WRITE_BITS(C, 8)

  // Write 16 bits with corrected endianess for 16 bit colours
  #define tft_Write_16(C) TFT_WRITE_BITS((C)<<8 | (C)>>8, 16)

  // Future option for transfer without wait
  #define tft_Write_16N(C) *_spi_mosi_dlen = 16-1;       \
                           *_spi_w = ((C)<<8 | (C)>>8); \
                           *_spi_cmd = SPI_USR;

  // Write 16 bits
  #define tft_Write_16S(C) TFT_WRITE_BITS(C, 16)

  // Write 32 bits
  #define tft_Write_32(C) TFT_WRITE_BITS(C, 32)

  // Write two address coordinates
  #define tft_Write_32C(C,D)  TFT_WRITE_BITS((uint16_t)((D)<<8 | (D)>>8)<<16 | (uint16_t)((C)<<8 | (C)>>8), 32)

  // Write same value twice
  #define tft_Write_32D(C) TFT_WRITE_BITS((uint16_t)((C)<<8 | (C)>>8)<<16 | (uint16_t)((C)<<8 | (C)>>8), 32)

//*/
#endif

////////////////////////////////////////////////////////////////////////////////////////
// Macros to read from display using SPI or software SPI
////////////////////////////////////////////////////////////////////////////////////////
#if !defined (TFT_PARALLEL_8_BIT)
  // Read from display using SPI or software SPI
  // Use a SPI read transfer
  #define tft_Read_8() spi.transfer(0)
#endif

// Concatenate a byte sequence A,B,C,D to CDAB, P is a uint8_t pointer
#define DAT8TO32(P) ( (uint32_t)P[0]<<8 | P[1] | P[2]<<24 | P[3]<<16 )

#endif // Header end
```

podaen, 2022-01-26:
To be sure, this is my print

http://www.lcdwiki.com/1.44inch_SPI_Module_ST7735S_SKU:MSP1443

podaen, 2022-01-26:
ok. I got it working now. I think I forgot to tft.init(); //TFT_eSPI in the setup(). here is my config

```
// Config for two ST7735 128 x 128 displays for Animated_Eyes example

#define ST7735_DRIVER     // Configure all registers

#define TFT_WIDTH  128
#define TFT_HEIGHT 128

// #define ST7735_INITB
// #define ST7735_GREENTAB
// #define ST7735_GREENTAB2
#define ST7735_GREENTAB3
 //#define ST7735_GREENTAB128    // For 128 x 128 display
// #define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
 //#define ST7735_REDTAB
//#define ST7735_BLACKTAB
// #define ST7735_REDTAB160x80   // For 160 x 80 display with 24 pixel offset

//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

//#define TFT_INVERSION_ON
//#define TFT_INVERSION_OFF

//// Generic ESP32 setup
#define TFT_MISO 19//
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    21
#define TFT_DC    2
#define TFT_RST   4  // Connect reset to ensure display initialises

// LED pin 3v
// SCK SCK 18
// SDA MOSI 23
// A0 DC 2
// RESET 4
// CS 21
// GND GND
// VCC pin 5v

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
```

it's faster than ever before. Thanks.

Bodmer, 2022-01-26:
If you use the User_Setup_Select.h header to decide which setup file to call up then you will not need to edit the User_Setup.h file.

Glad you have it working.

closed 2022-01-26 by Bodmer