Bodmer/TFT_eSPI #1578: TFT no longer functions on ESP32

pkendall64, 2022-01-26:
These ifdefs casue the TFT not to function any more on ESP32
[image]

I think they should probably be defined as 
[image]

Bodmer, 2022-01-26:
I am not able to reproduce this problem. There is a new function [initBus() ](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L520)that is called before that code, this initialises the pin states.

The code later is just for the ESP8266 and is needed in certain cricumstances.

In order to understand why you are experiencing this problem I need to know which version of the code you are using and ideally a simple sketch that demonstrates the problem.

pkendall64, 2022-01-26:
Sorry, yes 2.4.11 was working fine, upgrading to 2.4.21-25 causes the problem. We are using your library in the ExpressLRS codebase (http://gitbub.com/EspressLRS/ExpressLRS) for the Axis target.
Attached is a platformio project which demonstrates the problem. If you set the library version to 2.4.11 it works i.e. you get a  white screen, any newer version and the screen is blank.
[tfttest.zip](https://github.com/Bodmer/TFT_eSPI/files/7945220/tfttest.zip)

Bodmer, 2022-01-26:
I suspect this is down to the board package. In the Arduino board package you need v2.0.0 or later. It looks like adding support for the ESP32 C2 has introduced a problem with older board packages.

Bodmer, 2022-01-26:
I have just downgraded to board package 1.0.6 and it still works in the Arduino environment so ignore the las comment.

I do not use PlatformIO or ExpressLRS. Maybe [this is](https://github.com/Bodmer/TFT_eSPI/issues/1575#issuecomment-1022047439) relevant?

pkendall64, 2022-01-26:
Not sure if you saw this line

> EDIT: If I move the initBus() call to after the spi.begin() calls it works also!

I tried changing the platform as that other issue mentioned, but it would not link the application. Something about the .map file missing.

Bodmer, 2022-01-26:
Re. your comment: "If I move the initBus() call to after the spi.begin() calls it works also!" This implies that spi.begin is affecting control pins states. This suggests that you may be using default SPI port pins as control pins.

Which TFT pintergace GPIO pins are you using and what is the function allocation.

pkendall64, 2022-01-26:
The pins are defined as
```
#define GPIO_PIN_TFT_MOSI       16
#define GPIO_PIN_TFT_SCLK       17
#define GPIO_PIN_TFT_RST        0
#define GPIO_PIN_TFT_DC         12
#define GPIO_PIN_TFT_CS         22
#define GPIO_PIN_TFT_BL         15
```
and we also have `USE_HSPI_PORT` defined and the VSPI interface is used for the SX1280 radio chip.

Bodmer, 2022-01-26:
OK, in that case moving initBus() is the workaround solution.

I will see if it is possible to stop MISO being configured if it is not used.

pkendall64, 2022-01-26:
Awesome, thanks.

commit 42016590 (2022-01-26): Fix #1578
closed 2022-01-26 by Bodmer