Bodmer/TFT_eSPI #1581: Custom fonts loaded from SD Card (vs SPIFFS) contain artifacting

idorurez, 2022-01-27:
Hi @Bodmer, 

First off, this library has been a godsend, I have no idea where I would be without such an amazing tool.

I've finally got my SD Card working, and have been stuffing my fonts (and jpegs) onto it for testing. However, when I run `tft.loadFont(<font file>, SD);` and print with `tft.println(<text>)` the font looks anything but smooth; some equally placed "barbs" and whatnot around each character. When I switch back to loading the same custom font from SPIFFS, the font looks nice and as expected.

Can anyone confirm if they are seeing this too?

I am running ESP32-DEVKITC-V1

A little messy as I'm experimenting, but here is the current state with all the assets

https://github.com/idorurez/mini_weather_monitor/mini_weather.ino
https://github.com/idorurez/mini_weather_monitor/display.ino
https://github.com/idorurez/mini_weather_monitor/sdcard/   (contains all of the .vlw and jpegs I'm using)

Bodmer, 2022-01-27:
Which TFT are you using?
Is the SD card slot you are using on the TFT?

Bodmer, 2022-01-27:
The links above do not work but I found the repository.
Looks like you are using an ILI9486 at 27MHz. That may be too fast for that display type so try reduning the clock frequency.

idorurez, 2022-01-28:
Thanks @Bodmer, but turning it way down and other than a super slow display, there is no visual change; the artifacting bits remain.

To confirm, you're suggesting to turn down SPI_FREQUENCY correct?

idorurez, 2022-01-28:
I just found a similar post here: 

https://github.com/Bodmer/JPEGDecoder/issues/30

I just tried loading jpgs, and I'm having the same problems, little white blocks as you have mentioned could be corrupted images. I double checked the images and they're OK. I imagine this is related to my weird font issues. However in my case, I do have a separate CS pin I am using.

Bodmer, 2022-01-28:
I assume this is your setup file and connections?

This indicates there are two SPI busses being used. Try adding the following line to the setup file:

#define USE_HSPI_PORT

idorurez, 2022-01-29:
That worked! Thank you, and apologies I missed this in the repo!

https://github.com/Bodmer/TFT_eSPI/blob/8a2398451bc2027bed5391b85e4edce68fb6d1ce/User_Setup.h#L349

closed 2022-01-30 by Bodmer

Luis210380, 2022-05-04:
Hi Bodmer,

I'm connecting a ESP32 with a TFT display ILI9486 and im having the same problem (some white blocks in images loaded from SDcard) I'd tried to solve it using yours recommendations but the code doesn't work. I do have a separate CS pin I am using too. When I uncommented this line (//#define USE_HSPI_PORT) the display shows only black points in a low frequency. If I load images from internal mem of ESP, the cod does work correctly. In your opinion, What other cause is there for this problem?

Thanks for your kindly response

Bodmer, 2022-05-04:
@Luis210380 
Unfortunately I do not know what the problem is. It seems that the SD library does not play well with TFT_eSPI for some reason. 

The approach that some users have reported success with is to use separate busses for the SD card and display. That way they can be tested independantly with examples from each library.