Bodmer/TFT_eSPI #1588: Problems with driving ST7789 with esp32

QY7, 2022-01-28:
### Problem description
Hi, there, I'm currently using this library on my project, which uses **esp32** as the controller and **st7789** as the screen driver. I downloaded this library on platformio of vscode and modified the library settings according to the instruction but when I upload my program on my board, nothing shows up. 
I'm pretty sure the hardware connection is correct and the screen is fine because I have tried to display things on the screen with **LV_port_esp32** project, where the pin definition is the same as in the `User_Setup.h` settings.
### My project content
Here is my project structure
[image]
I have modified the `User_Setup.h` accordingly and here is my content. I have deleted most of the commented content to make it easier for reading.
```C
#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // ST7789 240 x 240
#define TFT_INVERSION_ON
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    27  // Data Command control pin
#define TFT_RST   25  // Reset pin (could connect to RST pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000
```
And my main.cpp file content is copied from the example project, which is
```C
#include <Arduino.h>
#include "driver/gpio.h"
#include "WiFi.h"
#include "TFT_eSPI.h"
// #define LED_BUILTIN 0
#define TFT_GREY 0x5AEB // New colour

TFT_eSPI tft = TFT_eSPI();  // Invoke library

void setup(void) {
  tft.init();
  tft.setRotation(2);
}

void loop() {
  
  // Fill screen with grey so we can see the effect of printing with and without 
  // a background colour defined
  tft.fillScreen(TFT_GREY);
  
  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line is there is room for the text with tft.print)
  tft.setCursor(0, 0, 2);
  // Set the font colour to be white with a black background, set text size multiplier to 1
  tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1);
  // We can now plot text on screen using the "print" class
  tft.println("Hello World!");
  
  // Set the font colour to be yellow with no background, set to font 7
  tft.setTextColor(TFT_YELLOW); tft.setTextFont(7);
  tft.println(1234.56);
  
  // Set the font colour to be red with black background, set to font 4
  tft.setTextColor(TFT_RED,TFT_BLACK);    tft.setTextFont(4);
  //tft.println(3735928559L, HEX); // Should print DEADBEEF

  // Set the font colour to be green with black background, set to font 4
  tft.setTextColor(TFT_GREEN,TFT_BLACK);
  tft.setTextFont(4);
  tft.println("Groop");
  tft.println("I implore thee,");

  // Change to font 2
  tft.setTextFont(2);
  tft.println("my foonting turlingdromes.");
  tft.println("And hooptiously drangle me");
  tft.println("with crinkly bindlewurdles,");
  // This next line is deliberately made too long for the display width to test
  // automatic text wrapping onto the next line
  tft.println("Or I will rend thee in the gobberwarts with my blurglecruncheon, see if I don't!");
  
  // Test some print formatting functions
  float fnumber = 123.45;
   // Set the font colour to be blue with no background, set to font 4
  tft.setTextColor(TFT_BLUE);    tft.setTextFont(4);
  tft.print("Float = "); tft.println(fnumber);           // Print floating point number
  tft.print("Binary = "); tft.println((int)fnumber, BIN); // Print as integer value in binary
  tft.print("Hexadecimal = "); tft.println((int)fnumber, HEX); // Print as integer number in Hexadecimal
  delay(1000);
  printf("%s","test\n");
}
```

### My efforts for debugging
[image]

In order to figure out why doesn't it work with this library, I have tried to probe the signal with my oscilloscope. RST is set low at startup and then goes high, VCC and GND is correct. Then, I saw signals on DC pin every 1s, but I didn't see any thing on SCLK and MOSI, I guess this might be the problem. However, I didn't find any clue any more. 
If you can help me, I would appreciate it.

Bodmer, 2022-01-28:
The display does not have a chip select and they do seem to be problematic. I think I have one so will try it.

In the meantime you can try adding this line to the setup file:
#define USE_HSPI_PORT

Also run the Read_User_Setup file an check the serial monitor output is correct.

Bodmer, 2022-01-28:
I tried mine and it is working fine.

This is the setup I used:
```
// ST7789 240 x 240 display with no chip select line

#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 240

#define TFT_MOSI 23 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 18
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   21  // LED back-light

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
#define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

```
This is the Read_User_Setup sketch output:
```
TFT_eSPI ver = 2.4.31
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 240
Display height = 240

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4

TFT_BL           = GPIO 21

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00

```

QY7, 2022-01-29:
Thank you for your timely reply. Unfortunately, I tried to copy your setup content directly and the screen still shows nothing, only black screen.
[image]

I have tried to modify the pin configuration in the **lv_port_esp32** project, which is shown below.
[image]
here is the pin configuration, which is the same as yours.
[image]

Is there any clue I can find? I still think it is the problem of the code. Besides, could you please provide me with the complete project you used?

Bodmer, 2022-01-29:
The main difference now is that you are using PlatformIO and I use the Arduino IDE. Users have reported success with TFT_eSPI so this must be an environment setup problem. Which version of the Arduino board support package are you using?

QY7, 2022-01-29:
Here is my platformIO configuration file
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = bodmer/TFT_eSPI@2.4.21
```

Bodmer, 2022-01-29:
Looks like you are using the Espressif IDF platform and TFT_eSPI will not run under that.

I loaded PlatformIO on my PC and created a simple project. It compiled,uploaded and ran OK. I added the setup information to the platformio.ini so that then it is not necessary to edit files in the TFT_eSPI library. You will need to adjust the pins but try this:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
lib_deps = bodmer/TFT_eSPI@^2.4.31

build_flags =
  -Os
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  -DUSER_SETUP_LOADED=1
  -DST7789_DRIVER=1
  -DTFT_WIDTH=240
  -DTFT_HEIGHT=240
  -DTFT_MOSI=23
  -DTFT_SCLK=18
  -DTFT_DC=2
  -DTFT_RST=4
  -DTFT_BL=21
  -DLOAD_GLCD=1
  -DLOAD_FONT2=1
  -DLOAD_FONT4=1
  -DLOAD_FONT6=1
  -DLOAD_FONT7=1
  -DLOAD_FONT8=1
  -DLOAD_GFXFF=1
  -DSMOOTH_FONT=1
  -DSPI_FREQUENCY=40000000
```

Bodmer, 2022-01-29:
I have looked deeper and the PlatformIO environment is using a fairly old version (1.0.0) of the Arduino ESP32 board package (when "platform = espressif32" is used) whereas the latest is 2.0.x

Using:
platform = https://github.com/platformio/platform-espressif32.git

Forces the use of 2.0.x so that is why this works in my setup.

To maintain compatibility with older versions I have patched the TFT_eSPI library and raised to 2.4.32. So it should now work with those older board packages when:
platform = espressif32
is used.

Thanks for raising this issue.

closed 2022-01-29 by Bodmer