Bodmer/TFT_eSPI #1594: New Release cannot compile.

nodoubtman, 2022-01-30:
Hello i have installed the new release and i got this error:

`TFT_eSPI.h:391:25: error: 'TFT_WIDTH' was not declared in this scope
   TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);

`

Can you help?
Thanks.
Marc.

Bodmer, 2022-01-31:
Can you tell me more about your setup as I cannot reproduce the problem.

I loaded a clean copy of TFT_eSPI 2.4.32, selected ESP32 Dev Module, with no edits to library I loaded and compiled the Colour_Test without error.

If TFT_WIDTH and TFT_HEIGHT are not defined then this implies a display driver has not been selected in your settings?

nodoubtman, 2022-01-31:
Seems fine now. I think i forgot to mention the display at the beginning of User_SEtup.h

Thanks.
Marc.

Bodmer, 2022-01-31:
Easily done! Thanks for letting me know.

closed 2022-01-31 by Bodmer