Bodmer/TFT_eSPI #1595: drawBmp SD

podaen, 2022-02-01:
I am trying to run  TFT_SD_BMP.zip from topic #95 . I am able to show the image but not correct.

I am calling it like this because without swapbytes nothing is shown

```
tft.setSwapBytes(true);
drawBmp("/data/clear-day.bmp", 0, 0);//, 16);
tft.setSwapBytes(false);
```

Than in the line below I get an error: expression must have a constant value.. The value of the variable w can not be used as a const.

uint8_t lineBuffer[w * 3];

This is the picture I get.

[image]

Whitout the drawBmp the sd and tft works (tested).

Bodmer, 2022-02-01:
This looks like exchanges with the SD card are somehow corrupting the screen.

Have you tried the example parrot image to check if the BMP image format is the problem?

What hardware are you using?

podaen, 2022-02-01:
I have multieple sd card reader, but I currently using this one.

[image]

Bodmer, 2022-02-01:
The first 3 cards have 5V to 3.3V buffers so are designed to connect to 5V boards like the UNO. You will need to power those from 3.3V only.

The last one might not have the buffer, but it might be other other side.

Ideally an unbuffered card would be better.

podaen, 2022-02-01:
That's true. That is wat I can tell out of my previous experiences too. Except the second from below works also on 3.3v. But with multiple spi devices it fails too, like the two above with a bufferchip.

Bodmer, 2022-02-01:
I just tested this sketch and it works fine. This uses the SD_MMC library with the SD card on different pins.
[TFT_SD_MMC_BMP.zip](https://github.com/Bodmer/TFT_eSPI/files/7982440/TFT_SD_MMC_BMP.zip)

podaen, 2022-02-01:
I just tested the first example TFT_SD_BMP.zip from topic #95 and it is still working. 

I did made some changes to my config before testing again see #1596 and did also solve this issue.

closed 2022-02-01 by podaen

Bodmer, 2022-02-01:
Great, thanks for letting me know.