Bodmer/TFT_eSPI #1598: RD pin in Parallel Mode

moreauftheobald, 2022-02-01:
Hello Back Again.

In order to have more free pin (and as i dont want to read data from my screen) i have not wired the RD pin of myscreen (this one is wired direcly to +3.3V and it work great)

Is it possible to update the labrary with the option #DEFINE rd = -1 ? Because for making the script compile i need to put a false pin into the setup file and i'm affraid that i could be a problem for me in the future (When i will use this pin as an DAC output).

Bodmer, 2022-02-01:
Either comment out the  #define TFT_RD line:

//#define TFT_RD xx

or set the pin to -1:

#define TFT_RD -1

closed 2022-02-01 by Bodmer

moreauftheobald, 2022-02-01:
When #define TFT_RD   is commented it dont work i got the following errors:
In file included from C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:20:
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'uint8_t TFT_eSPI::readByte()':
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:101:3: error: 'RD_L' was not declared in this scope
   RD_L;
   ^~~~
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:101:3: note: suggested alternative: 'WR_L'
   RD_L;
   ^~~~
   WR_L
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:106:3: error: 'RD_H' was not declared in this scope
   RD_H;
   ^~~~
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:106:3: note: suggested alternative: 'WR_H'
   RD_H;
   ^~~~
   WR_H
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::initBus()':
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:482:11: error: 'TFT_RD' was not declared in this scope
   pinMode(TFT_RD, OUTPUT);
           ^~~~~~
C:\Users\moreauf\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:482:11: note: suggested alternative: 'TFT_RED'
   pinMode(TFT_RD, OUTPUT);
           ^~~~~~
           TFT_RED

but when #define TFT_RD set to -1  it compile i'll test tomorrow to see if it work. Thanks for the quick answer

Bodmer, 2022-02-01:
Yes, just checked code for ESP32 and setting pin to -1 is correct approach.

I see an inconsistency here as leaving the pin definition commented out on an RP2040 would be OK. I will update for ESP32 at the next update.