Bodmer/TFT_eSPI #1601: Having trouble using TFT-eSPI with ESP32-S2 and ILI9341

malikbilal345, 2022-02-02:
Assalam-o-Alaikum everyone!
Have any one successfully tested ILI9341 and ESP32-S2?

I'm having trouble to run the examples on ESP32-S2, I configured the user setup as required but its not working. Earlier I have used alot this library with ESP32 and its just amazing. 

ESP32-S2 and ILI9341 is working fine for me with ADAFRUIT library but its not working with TFT-eSPI library. Anyone who can help me out? I shall be very thankful.

Regards.

Bodmer, 2022-02-03:
Use this as an example setup, note the HSPI port line:

https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h

You will need ESP board package 2.0.0 or later.

Run Read_User_Setup and look at serial output to verify settings are pickjed up by compiler.

homonto, 2022-02-03:
this problem is a regression from #1352 - I have the same problem: see my comment in #1352 please 
ESP32S2 was working until the newest update of the library

homonto, 2022-02-03:
in fast it is NOT only ILI - any display that I used so far (4 or 5 types) were ok till the newest upgrade to "version": "2.4.32"

Bodmer, 2022-02-04:
Comment out this line until I can resolve.
https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L16

Or use release version 2.4.2

homonto, 2022-02-04:
problem solved - it compiles now 
dirty but works!

thank you @Bodmer 
I think this shall stay open

malikbilal345, 2022-02-04:
Thanks a lot @Bodmer Sir, it really worked for me now.
Stay blessed!!!

Stewrat007, 2022-02-08:
Thank you for this temporary fix and for all your work on this Library , it is much appreciated
Stewrat

VEERAA2908, 2022-02-11:
Hi all,
I'm encountering compiling issues with Arduino IDE

Following is the error message i get

In file included from C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:20:0:
C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA(bool)':
C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:795:3: sorry, unimplemented: non-trivial designated initializers not supported
   };
   ^
C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:795:3: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:795:3: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:795:3: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:795:3: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Veeraa's PC\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:795:3: sorry, unimplemented: non-trivial designated initializers not supported
exit status 1

Is there any workaround for making this work?
Does it usually work in a Arduino IDE compiler?

Bodmer, 2022-02-11:
This came up in #1504 where the ESP32 board package was found to be the problem. Which board package version are you using? Latest is 2.0.2.

Bodmer, 2022-03-01:
The bug solution at the mement is to comment out[ this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L16).

Bodmer, 2022-03-20:
Fixed now. Finally spotted the typo!

commit c92c82be (2022-03-20): Fix #1601

A simple typo was the problem!
closed 2022-03-20 by Bodmer