Bodmer/TFT_eSPI #1612: Unable to compile for ESP8266 with latest master

Kotik82, 2022-02-04:
Hello, latest master doesnt compile for ESP8266.

\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::pushColor(uint16_t)':
\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3513:3: error: 'tft_Write_16N' was not declared in this scope; did you mean 'tft_Write_16S'?
 3513 |   tft_Write_16N(color);
      |   ^~~~~~~~~~~~~
      |   tft_Write_16S
exit status 1
Error compiling for board LOLIN(WEMOS) D1 mini Pro.

Kotik82, 2022-02-04:
Adding:

#ifndef tft_Write_16N(C)
  #define tft_Write_16N(C) tft_Write_16(C)
#endif

into TFT_eSPI_ESP8266.c resolves the issue.

Bodmer, 2022-02-06:
This should be fixed now.

Yes, smooth triangles and ellipses are planned.

The plan was to use the new wedgeLine function for the segments to improve appearance, the Smooth_Graphics_Demo includes a ring example but needs expanding to create a meter.

closed 2022-02-06 by Bodmer