Bodmer/TFT_eSPI #1614: Compile error " undefined reference to `TFT_eSPI::initDMA(bool)' " esp32

nbnaudb, 2022-02-05:
I have a problem with the latest version of the library. I have ESP32. I wanted to use the boing_ball example. When I try to compile, I get this message:

```
sketch\boing_ball.ino.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `TFT_eSPI::initDMA(bool)'
sketch\boing_ball.ino.cpp.o:(.literal._Z4loopv+0x8c): undefined reference to `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned int)'
sketch\boing_ball.ino.cpp.o: In function `setup()':
C:\Users\Piotrek\Documents\Arduino\libraries\TFT_eSPI-master\examples\DMA test\boing_ball/boing_ball.ino:66: undefined reference to `TFT_eSPI::initDMA(bool)'
sketch\boing_ball.ino.cpp.o: In function `loop()':
C:\Users\Piotrek\Documents\Arduino\libraries\TFT_eSPI-master\examples\DMA test\boing_ball/boing_ball.ino:163: undefined reference to `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned int)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Compilation error for MH ET LIVE ESP32DevKIT tiles.
```

this is what the User_Setup file looks like:

```
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
//#define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
//#define NUCLEO_64_TFT
//#define NUCLEO_144_TFT

// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTB_DATA_BUS

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
//#define ILI9341_DRIVER       // Generic driver for common displays
//#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER
//#define RM68140_DRIVER
//#define ST7796_DRIVER
//#define SSD1351_DRIVER
[... 279 lines cut ...]

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

```

Bodmer, 2022-02-05:
Unfortunately the ILI9486 is an 18bit colour display in SPI mode and DMA only supports 16 bit SPI displays. You can run the example without DMA. Comment out line 71 and line 163, uncomment line 166.

I will update the Readme to identify the 18 bit display types.

closed 2022-02-05 by Bodmer

summerfind, 2023-04-28:
Hi Bodmer.
Did you update the newer version of this library?
in my case, I try to use TTGO S3 to read analog value which seems can only use DMA to do it, any suggestions please?
Best Regards.
Summer

Bodmer, 2023-04-28:
@summerfind, the library does nt prevent DMA operations with other interfaces, e.g. analogue ADC values.

summerfind, 2023-04-28:
Thank you Bodmer. 
do you have a example of ADC-DMA read wave in Arduino IDE?
Best.