Bodmer/TFT_eSPI #1617: DMS example not compiled

unregistered33, 2022-02-05:
First, thanks for cool library! I have a litle problem with DMA test SpriteRotatingCube example. Core 2.0.1, esp32-Wrover-E 
Please help with it.
Error linking for board ESP32 Wrover Module
ld.exe: SpriteRotatingCube.cpp.o:(.literal._Z4loopv+0x1c): undefined reference to TFT_eSPI::dmaBusy()
ld.exe: SpriteRotatingCube.cpp.o:(.literal._Z4loopv+0x20): undefined reference to TFT_eSPI::pushImageDMA(int, int, int, int, unsigned short*, unsigned short*)
Build failed for project 'SpriteRotatingCube'
 
ld.exe: SpriteRotatingCube.cpp.o: in function setup()
SpriteRotatingCube.ino:158: undefined reference to TFT_eSPI  initDMA(bool)
 
ld.exe: SpriteRotatingCube.cpp.o: in function loop()
SpriteRotatingCube.ino:228: undefined reference to TFT_eSPI  dmaBusy()
ld.exe: C:\Workspace\VisualMicro\Projects\libraries\TFT_eSPI-master\examples\DMA test\SpriteRotatingCube\SpriteRotatingCube.ino:229: undefined reference to TFT_eSPI::pushImageDMA(int, int, int, int, unsigned short*, unsigned short*)
 
collect2.exe*: error: ld returned 1 exit status

unregistered33, 2022-02-06:
im using ili9486 with SPI. Other examples (not DMA) works fine

Bodmer, 2022-02-07:
The Ili9486 uses 18 bit colour in SPI mode and this is not supported by DMA. I have updated the ReadMe to clarify.

closed 2022-02-07 by Bodmer