Bodmer/TFT_eSPI #1626: Smooth graphics

videobuff, 2022-02-08:
Hi Bodmer, More and more I get the hang of your library.  I even got the touch working on a WT32-SCO1 display.
Now i wanted to make use of the smooth graphics which you added a couple of days ago
Started with the examples but hit problems straight away.
In short:
Both of the examples do not compile on my machine. I use arduino 1.8.19 and TFT_eSPI version 2.4.32.
The demo version start with:  
`smooth_graphics:42:7: error: 'class TFT_eSPI' has no member named 'drawSpot'; did you mean 'drawRect'?`
and the smooth clock throws
`smooth_graphics:151:8: error: 'class TFT_eSprite' has no member named 'drawWideLine'; did you mean 'drawLine'?`
Best reards
Erik

Bodmer, 2022-02-08:
You may have picked up a copy that was being updated. Please try latest 2.4.35.

videobuff, 2022-02-09:
That was it. Arduino still showed 2.4.32 as the latest but a download from GitHub sorted it out. I use it on a WT32-SCO1 display which is a wrover, and although everything works, i get a few warnings. See below.
Thanks again for a wonderful library..
Erik

```
In file included from /Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:65,
                 from /Users/erikschott/Documents/Arduino/clock_smooth_graphics/clock_smooth_graphics.ino:15:
/Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:41:22: warning: extra tokens at end of #ifndef directive
 #ifndef tft_Write_16N(C)
                      ^
In file included from /Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:65,
                 from /Users/erikschott/Documents/Arduino/clock_smooth_graphics/clock_smooth_graphics.ino:15:
/Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:548: warning: "tft_Write_16N" redefined
   #define tft_Write_16N(C) *_spi_mosi_dlen = 16-1;       \
 
In file included from /Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:65,
                 from /Users/erikschott/Documents/Arduino/clock_smooth_graphics/clock_smooth_graphics.ino:15:
/Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:42: note: this is the location of the previous definition
   #define tft_Write_16N(C) tft_Write_16(C)
```

closed 2022-02-09 by videobuff

Bodmer, 2022-02-09:
Great.

I have fixed the warnings inthe latest version.