Bodmer/TFT_eSPI #1627: drawCentreString() draws at an negative offset

dkurok, 2022-02-09:
I use a 320x240 ILI9341-driven TFT on an ESP32.
``` 
tft.setRotation(1);
tft.setTextColor(TFT_SKYBLUE, TFT_DARKGREY);
tft.setTextFont(7);
tft.setTextSize(1);
int xOffset = 0;
int yOffset = 0;
tft.drawCentreString("0", 0+xOffset, 0+yOffset, 7);
```
This draws the "0" only half; the first half of it is left outside the screen (so it seems the x-Postion 0 is outside the screen).
I've got to add ~20 (using xOffset=20 above) for getting it OK. In Y-direction there is no such problem.
If I set xOffset to 320, the first half of the "0" is even visible on the right top corner of the screen.
Do I have to do some calibration for the TFT itself? (I know that I've got to do it for the also integrated touch; but that is not used here)

Bodmer, 2022-02-09:
The top left corner of the screen is coordinate 0,0 so the rendering is as expected. The drawCentreString function calculates the top centre position of the string and puts this point at the specified coordinate. Thus a 0,0 you will see half a single character.

drawCentreString is a historic function and is deprecated, it is better to use use setTextDatum(), see examples.

Bodmer, 2022-02-09:
See FreeFontDemo to understand how the text datum functions work.

closed 2022-02-09 by Bodmer