Bodmer/TFT_eSPI #1630: ESP8266 - Only Backlight No Graphics on ST7789 TFT Display

audreylim, 2022-02-10:
Hi!

I'm trying to get my TFT display working, but all I see is the backlight and no graphics. I uploaded this code https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Gradient_Fill/Gradient_Fill.ino to my ESP8266 from the Arduino IDE.

I got my ESP from here https://www.amazon.com/HiLetgo-Internet-Development-Wireless-Micropython/dp/B010O1G1ES/ref=asc_df_B010O1G1ES/?tag=hyprod-20&linkCode=df0&hvadid=309818716690&hvpos=&hvnetw=g&hvrand=13377053487764932588&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9067609&hvtargid=pla-361876718784&th=1
and my 1.3" TFT 240x240 SPI ST7789
and my TFT from here https://www.amazon.com/MakerFocus-Display-1-3inch-Interface-Routines/dp/B07P9X3L7M

My PIN connections are:
GND - GND
VCC - 3v3
SCL - D5
SDA - D7
RES - D2
DC - D1
Reference: https://simple-circuit.com/esp8266-nodemcu-st7789-tft-ips-display/

I've uploaded the code, unplugged and plugged the usb, held on to the reset button, but all I see is the backlight. Is there something I am missing?

Thank you!

[image]
[image]

sayanee, 2022-02-10:
@audreylim It seems like you are connecting to the 2 I2C pins `SCL - D5, SDA - D7` when it should be the 4 SPI pins. Re-wire again?

[image]

Bodmer, 2022-02-10:
GPIO 5 and GPIO 7 are the correct pins.

Run the Read_User_Setup diagnostic sketch to check you have correctly edited the user setup file. Look at the serial output to check compiler is picking up correct pins and TFT driver. Post output here if you are not sure.

Bodmer, 2022-02-10:
You must solder the header pins to the display as well!

audreylim, 2022-02-10:
Thanks @Bodmer , @sayanee 

I uploaded the Read_User_Setup and got this:

[image]

Bodmer, 2022-02-10:
You need to set the monitor window baud rate to 115200

Bodmer, 2022-02-10:
I made a mistake above, I meant pins D5 and D7 are correct. These map to GPIO 14 and GPIO13.

Bodmer, 2022-02-10:
Great, glad it is working for you.