Bodmer/TFT_eSPI #1631: "tft_Write_16N redefined" with TFT_eSPI master using the RP2040 (Raspberry Pi Pico)

cbrewer314, 2022-02-11:
Hello,
I recently re-installed the TFT_eSPI library in my Arduino environment with the 'master' branch and now I am seeing the following error:
`
/home/chris/Arduino/libraries/TFT_eSPI-master/Processors/TFT_eSPI_RP2040.h:305: warning: "tft_Write_16N" redefined
  305 |       #define tft_Write_16N(C)    while (!spi_is_writable(SPI_X)){}; spi_get_hw(SPI_X)->dr = (uint32_t)(C)
      | 
/home/chris/Arduino/libraries/TFT_eSPI-master/Processors/TFT_eSPI_RP2040.h:111: note: this is the location of the previous definition
  111 |   #define tft_Write_16N tft_Write_16
      | 
`
This is with the 'Raspberry Pi Pico' board and the 'earlephilhower' core.
Unfortunately I did not determine the previous version for the TFT-eSPI library that I was using before I deleted the previous one...

Bodmer, 2022-02-12:
The library should still compile and run as it is a warning. I will solve this.

cbrewer314, 2022-02-13:
> Fixed in v2.4.37

Excellent, thank you!