Bodmer/TFT_eSPI #1632: Misaligned 4bit sprites when width is odd

raomin, 2022-02-12:
I created a 4bit sprite with a BMP file with a odd width.
When pushed to tft, the image is like stretched, misaligned.
It looks like the code does not handle well the fact that the last byte of the line codes only for one pixel...

No issue with a width even, or by removing 1 on the width of `pushImage`.

my image.h 
```c
// width is 37, height is 52
static const uint8_t mapTemperature[1040] PROGMEM = {
...
```

But with this, the sprite lines are misaligned:
```c
      sprite.createSprite(37,52);
      sprite.setColorDepth(4);
      sprite.createPalette(paletteTemperature);
      sprite.pushImage(0,0,37,52,(uint16_t*) mapTemperature);
``` 

With this it's ok:
```c
      sprite.createSprite(37,52);
      sprite.setColorDepth(4);
      sprite.createPalette(paletteTemperature);
      sprite.pushImage(0,0,36,52,(uint16_t*) mapTemperature); // <<<-- changed from 37 to 36
```

Bodmer, 2022-02-12:
I suspect this is due to an incorrect assumption on how the sprite image data is stored. For an odd width the line consists of a whole number of bytes with the last LS nibble being padding. So a 3 pixel wide sprite has 2 bytes per line.

As I do not know what is in your array I am unable to confirm this.

raomin, 2022-02-12:
take this one for example: https://gist.github.com/raomin/f38de71a44570c03589422cbe3e76f6e
Just used `bmp2array4bit.py` to do it.

see with 
```c
      sprite.createSprite(37,52);
      sprite.setColorDepth(4);
      sprite.createPalette(paletteTemperature);
      sprite.pushImage(0,0,37,52,(uint16_t*) mapTemperature);

```

raomin, 2022-02-12:
Ok, I would recommend to add a comment in bmp2array4bit.py and its readme.
I spent quite some time struggling with this as might do others, 50% of the time...

Bodmer, 2022-02-12:
Yes, post an issue on that repository:
https://github.com/sparkfun/BMPtoArray

raomin, 2022-02-12:
Now, looking at it, the python scrip to me is correct in handling 2 pixels as on byte even if they are in separated rows.
Considering a 3x2 image:
```
1 2 3
4 5 6
```
I would assume to get a 3 byte map for the image, not a 4 byte one with padding. At least that's how [BMPtoArray](https://github.com/sparkfun/BMPtoArray) does it; it handles the image as one stream of pixel.

I'm curious, why do you say _the last LS nibble being padding_, is it a kind of (suboptimal) standard?
If so, I guess I can suggest an option in [BMPtoArray](https://github.com/sparkfun/BMPtoArray)'s repo.

Bodmer, 2022-02-13:
There appear several things wrong with the python script. The example in the [provided link](https://gist.github.com/raomin/f38de71a44570c03589422cbe3e76f6e#file-temperature-h-L11) is clearly not a 37 x 52 image, since only 18 x 52 bytes are listed. So 36 x 52 pixels of 4 nits each. The array size of 1040 is also wrong. It should be ((37 * 52) / 2) = 962 bytes. So for some reason the array size is 78 bytes too big.

Padding is a common technique used in images, it is for example used in the BMP format. The reason is that it simplifies pixel indexing code and the overhead of padding is typically negligible.