Bodmer/TFT_eSPI #1640: Per-project display setup instead of editing always User_setup.h

homonto, 2022-02-15:
Hi, 
I know I can uncomment 1 line and comment the other one in the "User_Setup_Select.h" file, to have my setup included (per display).
But, I am thinking about more robust way.
Why?
Since I have few displays and projects, it happened to me already few times to compile the project with "wrong" display.
Outcome? Nothing on display (of course).
Is there any possibility to do something like this in my ino file:
#define MY_DISPLAY_1
and then, in the file "User_Setup_Select.h" put something like this:
#ifdef MY_DISPLAY_1
#indclude <MY_DISPLAY_1.h>
#endif

I actually tried it but apparently it is not working - complaining too much - apparently such directive is messing around

If such mechanism exists, I would never ever mess up with my projects.

Thank you

Bodmer, 2022-02-15:
The Arduino IDE does not allow sketch #defines to be visible to a library  hence the use of setup files.

If you use PlatformIO as the IDE then you can have project based setups in the platformio.ini files  see tools folder.

One possibility with the Arduino IDE is to generate a board type specific to a project but there are no tools to do this, and it would still be necessary to select the right board in the IDE when a different setup is wanted.

It might be possible to write an add in for the IDE that behaves as as project manager but I have not tried that.

Another option is to allow a sketch to detect the setup in use and generate a compile time error if it is wrong. That would be an easy to add feature.

homonto, 2022-02-15:
thank you Bodmer - it seems Arduino is for lamers so I am one of them and I need to stay there as Platformio has other things that are not working i.e. S2 is still not there

homonto, 2022-03-08:
btw I made it working by using some macros - in case someone is interested in how (maybe dirty code but... working).
when I use ST7789 display I define in my ino file:
`#define ST7789_LCD`

for ILI I use:
`#define ILI9341_LCD`

I actually at this moment of time use only these two.

Then in  void setup():
```
  #if defined(ILI9341_DRIVER) and defined(ST7789_DRIVER)
//the above comes from your setup files where you define which driver is in use
    #error "2 LCD drivers defined - choose only ONE please"
    #error "in file User_Setup_Select.h "
  #endif

  #ifdef ILI9341_LCD
    #ifndef ILI9341_DRIVER
      #error "wrong display configured - change the line in User_Setup_Select.h file accordingly"
      #error "in file User_Setup_Select.h "
    #endif
  #endif
```

so thank you for the hints