Bodmer/TFT_eSPI #1641: How to reset a screen?

raomin, 2022-02-15:
Hi @Bodmer,

I have a screen ili9341 connected to an ESP32. I implemented a timeout that slowly shuts the screen down (LED PWM to 0 +   `tft.writecommand(0x10); delay(5);`).
Pressing on a button slowly wakes it up (`tft.writecommand(0x11); delay(120)`; LED PWM to 255)

Overall all is OK.
But sometimes, I wake up on a blank screen. Like if the SPI connection is down, or unsynced. I have to manually reset the ESP32 to fix the issue.
I've looked at other similar issues, they linked it to EMC issues but I don't have any strong current, only few ws2811 leds...

I have 2 questions:
- Do you think about anything that could cause this issue? Or what could avoid it? (I have put the SPI frequency to 27MHz just in case but still happening...)
- How could I reset the screen (instead of just LED off)? I presume this would help preventing this issue.

Thanks for your awesome work and support!

Bodmer, 2022-02-16:
You can call tft.init(); to refresh the TFT driver.

I think you have the delays the wrong way around. The data sheet says:
```
It will be necessary to wait 120msec after sending Sleep In command (when in Sleep Out mode)
before Sleep Out command can be sent.
```
In other words the display must stay at least 120ms in Sleep mode.

Bodmer, 2022-02-16:
I see the data sheet also saysL
```
 It will be necessary to wait 120msec after sending Sleep Out command (when in Sleep
In Mode) before Sleep In command can be sent.
```
This implies you must wait 120ms between both sleep in and sleep out commands. Not sure if this is part of the problem though.

closed 2022-02-16 by Bodmer