Bodmer/TFT_eSPI #1643: Wrong touch coordinates

unregistered33, 2022-02-17:
ili9486 spi, esp32. 
I try the example "Touch Controller Demo" from the library and get a strange result. The point is drawn not at the point of touch, but to the side, and in different places on the screen, the distance from the touch point to the point on the screen is different. I see the dependence of distances on a point on the screen, but I can’t understand what the problem is. I do pre-calibration all the time.

https://user-images.githubusercontent.com/1637078/154388270-06c0c16d-5d66-4092-9b07-1914f72716e6.mp4

Bodmer, 2022-02-17:
I suspect that the screen is particularly sensitive. It appears that on some touch screen the response is slow and the detected touch point moves slowly from one end of the screen. This causes calibration to detect the wrong limits.

You may be able to see this effect in the readings reported by the raw touch test diagnostic sketch when touching screen corners.

You could try reducing the touch SPI clock rate to 1MHz so the touch value has more time to settle.

Also try raising the touch threshold to 600 here:
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Touch.cpp#L162

If it is raised too high the corner touches may not be detected, you can see the Z value range in the test diagnostic sketch.

unregistered33, 2022-02-18:
Unfortunately the advice didn't help. Apparently the touchscreen or touchcontroller is faulty

closed 2022-02-24 by Bodmer