Bodmer/TFT_eSPI #1650: Dose ESP32 S2 support 8080 8bit parallel LCD

uldara1, 2022-02-19:
- [ ] Hi i try to configure TFT_eSPI with ST7796 parallel interface after upload code display not display anything
does my configure correct ? pls check thank
 

`
#define ST7796_DRIVER
#define TFT_WIDTH 320
#define TFT_HEIGHT 320 

#define TFT_PARALLEL_8_BIT
#define TFT_CS -1  // Chip select control pin (library pulls permanently low
#define TFT_DC 38  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST 37 // Reset pin, toggles on startup
#define TFT_WR 39 // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD -1 // Read strobe control pin
#define TFT_D0 36 // Must use pins in the range 0-31 for the data bus
#define TFT_D1 35 // so a single register write sets/clears all bits.
#define TFT_D2 34 // Pins can be randomly assigned, this does not affect
#define TFT_D3 33 // TFT screen update performance.
#define TFT_D4 21
#define TFT_D5 18
#define TFT_D6 17
#define TFT_D7 16
`

Bodmer, 2022-02-19:
At the moment the library has not been tested or even looked at to see if 8 bit parallel can be used with the ESP32  S2.

closed 2022-02-19 by Bodmer

uldara1, 2022-02-20:
Ok thank you
I tested with ESP32 S2 but it not work yet