Bodmer/TFT_eSPI #1651: 'class TFT_eSPI' has no member named 'getTouch'

MercedesAMGSantiago, 2022-02-19:
Hi, I am getting this error while compiling a code for ESP8266:

**Multiple libraries were found for "TFT_eSPI.h"
 Used: /home/davidsantiago/Arduino/libraries/TFT_eSPI
 Not used: /home/davidsantiago/Downloads/arduino-1.8.13-linux64/arduino-1.8.13/libraries/TFT_eSPI-master
 Not used: /home/davidsantiago/Arduino/libraries/TFT_eSPI-master
exit status 1
'class TFT_eSPI' has no member named 'getTouch'**

I already put the correct pins on the User_setup.h, can anyone help? Thanks.

Bodmer, 2022-02-20:
Run the Read_User_Setup sketch and check the serial monitor output. This will tell you if the correct setup is being used and that the TOUCH_CS pin is defined.

Bodmer, 2022-02-24:
Assumed resolved.

closed 2022-02-24 by Bodmer