Bodmer/TFT_eSPI #1653: OLED ST1351: Smooth fonts can be smoother?

audreylim, 2022-02-20:
Hello, I'm trying to display numbers on my 1.5inch OLED ST1351 128x128 (with ESP8266). I wanted the text to look as sharp as this:

[image]

As advertised: https://www.hawkusa.com/manufacturers/newhaven/displays/nhd-15-128128ugc3?gclid=CjwKCAiA6seQBhAfEiwAvPqu1xuJyPlAAslXiiE0YohiaL7sGPuy0Q19w6c7ZqEvOC4Qu92xEk6CWRoCncIQAvD_BwE

Instead, I got this:

[image]

I'm running this code: https://github.com/Bodmer/TFT_eSPI/blob/11b006e1d926b84508d257fdac9d7e90cabfcd5f/examples/Smooth%20Fonts/SPIFFS/Print_Smooth_Font/Print_Smooth_Font.ino

This code is much sharper than the other fonts like yellowtail etc, but it's still not sharp as the one in the first image. Am I missing something or is the sharpness constrained by the OLED model?

audreylim, 2022-02-20:
Hmm so this font looks pretty smooth on my LCD https://github.com/Bodmer/TFT_eSPI/blob/master/examples/160%20x%20128/RLE_Font_test/RLE_Font_test.ino

Only problem is that I can't make the text size smaller. 

I also cannot use the letter "S" with font `6`, or the whole thing doesn't render.
```
   tft.drawString("S123", 0, 0, 4);
```
Looks like this is an issue:
https://github.com/Bodmer/TFT_eSPI/issues/965 

Any solutions to still get smooth fonts and be able to make the the text smaller?

Bodmer, 2022-02-20:
The [fonts with numbers 6,7 and 8 are numerals only plus a few others ](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L299-L301)for numeric/time values of the screen. These fonts were built for the UNO originally which was very FLASH constrained.

I am not sure what you mean by "sharpness". The smooth fonts use anti-aliasing to reduce pixel resolution jaggedness.

Bodmer, 2022-02-20:
If your are not getting the font you expet it may be becuase you have not created a SPIFFS partition and uploaded the files.

If you have enough FLASH then you can [use these array based examples](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Smooth%20Fonts/FLASH_Array) since then you do not have to configure and upload the SPIFFS files.

audreylim, 2022-02-20:
Hmm, I uploaded the examples again and they are anti-aliased, thank you.

Is there a way to adjust the size of this font? https://github.com/Bodmer/TFT_eSPI/blob/master/examples/160%20x%20128/RLE_Font_test/RLE_Font_test.ino

audreylim, 2022-02-20:
[image]
[image]

I'm actually getting these when running this code: https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Smooth%20Fonts/FLASH_Array/Font_Demo_1_Array

The large font has noise at the back and the small font is not rendered cleanly.

Bodmer, 2022-02-20:
Looks like screen rotation does not work on that screen driver. Try rotation 0 only (default). That OLED driver was submitted by a user, I don't have one of those screens, so I am unable to test :-(

Bodmer, 2022-02-21:
I had a look at this and see I changed the pixel drawing code recently and introduced a bug.

I will upload a new fixed version.

Bodmer, 2022-02-21:
New version is 2.4.40, please test as I am unable to do so.

closed 2022-02-21 by Bodmer

audreylim, 2022-02-21:
[image]
[image]
[image]

Much better although the last image still seems buggy.

audreylim, 2022-02-21:
I just realized it's intentional https://github.com/Bodmer/TFT_eSPI/blob/c11ecd6932d50cdd63260e346a8b10eb519c7a69/examples/Smooth%20Fonts/FLASH_Array/Font_Demo_1_Array/Font_Demo_1_Array.ino#L107. 

Thanks! It works fine.

Bodmer, 2022-02-21:
Thanks for testing and reporting this problem.