Bodmer/TFT_eSPI #1658: 16bpp sprite: PushImage followed by PushSprite gives incorrect colors

konkrog, 2022-02-21:
Loading the image directly on the tft using tft.pushImage() works fine...

However, when trying to first load the image into a 16bit sprite using sprite.pushImage() then pushing the sprite to the tft using sprite.pushSprite() the colors are incorrect.

The following code gives incorrect colours:
`TFT_eSprite splash = TFT_eSprite(&tft);

  splash.createSprite(128,100);

  splash.fillSprite(TFT_BLACK);

  const int logo_name_spacing = 5;
  // calculate position of logo upper left hand corner
  int x = (splash.width()-image_width_elogo)/2;
  int y = (splash.height()-image_height_elogo)/2;

  // logo to center of screen
  splash.pushImage(x, y, image_width_elogo, image_height_elogo, image_data_elogo );
  splash.pushSprite(0,0);`

Example of incorrect colors:
[image]

The issue can be corrected by adding a call to setSwapBytes(true) before calling pushImage(). The following code works:

`TFT_eSprite splash = TFT_eSprite(&tft);
  splash.createSprite(128,100);
  splash.fillSprite(TFT_BLACK);

  const int logo_name_spacing = 5;
  // calculate position of logo upper left hand corner
  int x = (splash.width()-image_width_elogo)/2;
  int y = (splash.height()-image_height_elogo)/2;

  // logo to center of screen
  splash.setSwapBytes(true);  
  splash.pushImage(x, y, image_width_elogo, image_height_elogo, image_data_elogo );
  splash.pushSprite(0,0);`

Example of correct colors:
[image]

konkrog, 2022-02-21:
The sprite should possibly inherit the swap byte property from the tft... so

`splash.setSwapBytes(tft.getSwapBytes()); // Sprite should inherit the swapBytes property from tft...(?)`

might be a better temporary correction that the one I proposed above.

Bodmer, 2022-02-22:
The library is operating as intended. Control of byte swapping per sprite is important, since this permits byte swapped images to be pasted into sprites.

Ideally the source file should not be byte swapped, but that is not always under the control of the user.

closed 2022-02-22 by Bodmer

konkrog, 2022-02-22:
Ok.... I don't really understand why pushing the same image directly to the display should be different from pushing it via a sprite...(?)... I understand that the API will allow the user to control the byteswapping... but does it not make sense that the sprite upon initialization inherits the byte swapping properties of the display?

Bodmer, 2022-02-22:
Yes, unfortunatley it is confusing. The reason is that sprites are notionally stored byte swapped, so that means bytes can be sent direct to a SPI port on the ESP32 via DMA without having to swap them in transit. Some processors can be configured to automatically byte swap data sent to the SPI port, but the ESP32 cannot do that in hardware. To add to the confusion different processors have different endianess, which affects how data is stored!