Bodmer/TFT_eSPI #1660: HX8357D SPI Driver has inverted text and odd colors

Pconti31, 2022-02-23:
I'm using Adafruit's 3.5 feather tft display spi HX8357D and Adafruit ESP32 Feather.  
The combination works fine with your release TFT_eSPI-2.3.4 but when I ran some test programs using 2.3.4.32 text is reversed and colors not quite inverted but different.  Placement of text and icons are also inverted.
Note that my test programs don't set rotation.  
When I call for rotation, I tested  for landscape 1,3,5,7 and found 7 mostly worked but colors were still wrong.

I replaced HX8357D_Rotation.h with TFT_eSPI-2.3.4/TFT_Drivers/HX8357D_Rotation.h and my original unmodified test programs all worked fine.

Bodmer, 2022-02-24:
Thanks for reporting this. I do not have one of those displays, but [changed the rotation code as a pull request](https://github.com/Bodmer/TFT_eSPI/commit/f3135aea29d66701c8012f92e83b2846bbc9f9d5) from a user.

Looks like that was not a good idea...

Can you test all rotations with the original code and verify they work, then to be clear post the required rotation code here.

Bodmer, 2022-02-24:
I have reverted the change, please test with master from Github.

Pconti31, 2022-02-24:
@Bodmer I tested against master and cycled through, not using set rotation, then setting rotations of 0,1,2 and 3. All worked fine.
Thank You!

closed 2022-02-24 by Pconti31

Bodmer, 2022-02-24:
Thanks for letting me know.