Bodmer/TFT_eSPI #1669: A color related bug using DMA.

QY7, 2022-02-27:
Hi, there. I'm currently using LVGL together with TFT-eSPI. I have set the `#define LV_COLOR_16_SWAP   1` and the color display is normal when I do not use DMA. However, when I switch to use DMA to transfer to info, the color seems wierd and I have to define `LV_COLOR_16_SWAP` to 0.
I am wondering why the two approaches need different color config, am I missing something? thank you.

BTW, I am using lvgl8.0 and platformIO on VScode. And my board is ESP32Dev. Thank you in advance.

Bodmer, 2022-02-27:
This seems to be a compatibility problem or issue with the impementation of the display update code rather than a bug. 

Typically I use pushImage(..) or pushImageDMA(...) to copy pixel blocks to the display, and these are compatible in terms of endianess.

You will need to double buffer the data so that the screem image buffer is not being updated by LVGL while DMA is still in progress. Similarly if LVGL responds quickly with the next display buffer ou will need to wait before initiating another DMA (pushImageDMA does this for you and you can specify a second buffer in the function and then data will be double buffered.

I assume you have run the TFT_eSPI examples without a problem. Without looking at your display upate code I cannot advise further.

Bodmer, 2022-03-01:
Post your display update code and I will check it.  Since I suspect this is an implementation issue I will close for now.

closed 2022-03-01 by Bodmer