Bodmer/TFT_eSPI #1670: Request to add support for 5" SSD1963 from TSD display.

hackeybabel, 2022-02-27:
[5.0LCD (1).zip](https://github.com/Bodmer/TFT_eSPI/files/8148146/5.0LCD.1.zip)

I am trying to use this display with stm32f405rgt6 i tried all the versions in your library but with no success. I would be very helpfull if you can add this display support to your library.

I am attaching init code i got for this display. 

My connections are as follows 

#define TFT_CS   PB0  
#define TFT_DC   PB1  
#define TFT_RST  PB4  

#define TFT_WR    PB5  
#define TFT_RD    PB6 

#define TFT_D0   PA0 
#define TFT_D1   PA1  
#define TFT_D2   PA2 
#define TFT_D3   PA3  
#define TFT_D4   PA4
#define TFT_D5   PA5
#define TFT_D6   PA6
#define TFT_D7   PA7

[image]
[image]

Bodmer, 2022-02-27:
Support has already been added for SSD1963 controllers.

The first step is to get the code the supplier provided working and running with a full 16 bit interface.  The display appears to be configured for a 16 bit interface only. You will need to see if there are configuration pins on the display to convert to 8 bit. If so then convert both the display and supplier provided code to 8 bit and test again.

Note: TFT_eSPI only supports 8 bit parallel interfaces. Ideally keep the bus connections short (ideally <100mm) to prevent signal integrity issues.

Next you will need to examine the init values in the provided and edit [the setup file](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/SSD1963_Init.h#L3-L103) in TFT_eSPI.

In view of the large number of potential errors and problems with this type of display and connection setup I am not able to help further.

closed 2022-02-27 by Bodmer

Bodmer, 2022-03-03:
I have updated the info above, since the links were not quite right.

Bodmer, 2022-03-03:
I have checked the data sheet and see that you need to set the CONF pin to logic 1 (8080 mode) as the library uses RD and WR signals. Check the state on your board, or looking at the supplied code will indicate this (signals might be named for example E and R/W in 6800 mode)).

The display is commanded into 8 bit mode during init with an 8 bit bus, so there are no configuration pins for that. The data sheet indicates the other Dx pins should be connected to 0V but I am not sure if that is essential, but otherwise they will "float" which is bad for some chips.