Bodmer/TFT_eSPI #1679: screen turn white and flashy when wifi.begin

infrafast, 2022-03-02:
Hello, 

I observed  that when I call for wifi.begin  the screen turn white , like if it has lost the initialisation.

Equally, if I call for wifi WiFi.mode(WIFI_AP_STA);  and  WiFi.softAP(...) this is even worse as the screen become white but flashy

I initially suspected a power peak as I am using also an ethernet w5500 chip, but I have removed it, still same problem when I have only the tft sceen attached and the peak goes to 380mA having a board able to deliver 400mA

Then I tought this could be momory but I have ~20K of heap before the call and after the call to wifi.begin

Then I tried to use different value for #define SPI_FREQUENCY , still problem persists.

I've tested with different ILI9341 screen and this happens quite randomly

I arrive to the point where I don't know what to check.... 

Here is an extract of my user_setup:

#define ILI9341_DRIVER

#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D1  // Data Command control pin
#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters

#define SPI_FREQUENCY  27000000           //works ok
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

#define SUPPORT_TRANSACTIONS

dev environnement : platform IO

PLATFORM: Espressif 8266 (3.2.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 

Thanks a lot for having a look to this issue.

Bodmer, 2022-03-02:
This sounds like a peak power demand problem. Does the ESP8266 reboot repeatedly when this happens. Things to try:
1. Add a 470uF decoupling capacitor at the ESP8266 3.3V power pins.
2. Try a different power source.
3. Check you have short 0V  connections between parts.
4. Change sketch to reduce memory consumption as a test.
5. call tft.init() again after WiFi operations to re-inutialise the TFT, however this does not fix the root cause and that cause may have other consequences.
Since WiFi is not part of thus libraries functionality and that is the fault trigger then I will close, post back if you solve it to help others.

closed 2022-03-02 by Bodmer

infrafast, 2022-03-02:
Hello thanks a lot for having checked my concern. 

 _Does the ESP8266 reboot repeatedly when this happens_ -> no the operation continues normally

calling tft.init()  after issue happens sometimes works sometime not.

I will further investigate and post here if I find root cause.

Bodmer, 2022-03-02:
If the WS5500 uses SPI, then check for pins conflicts. The WS5500 chip select must be a different pin. If the WS5500 uses interrupt then that might cause issues too. When you disconnected it did you also remove the associated software elements from the sketch?

infrafast, 2022-03-02:
indeed, I also thought about SPI conflict but both the screen and eth.chip have their own CS, so we can exclude HW wiring.
Regarding software, it may be an interrupt however, if the wifi.begin() pass correctly (which happens most of the time), then the tft AND w5500 co-work together without any problem,. Issue happens when wifi kick in....

Bodmer, 2022-03-02:
I have just run this example on an ESP8266:
https://github.com/Bodmer/TJpg_Decoder/tree/master/examples/LittleFS/LittleFS_Web_Jpg

It connected to WiFI, got the image from the server and displayed it on the TFT. Does it work for you?

Note: You need to allocate space for LittleFS and also enter you WiFi SSID and password in the sketch.

Note that it uses the ESP8266 specific WiFi libraries.

My TFT setup file (I use a ILI9225):
```
#define ILI9225_DRIVER

#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  80000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

#define SUPPORT_TRANSACTIONS

```

Serial monitor output:
```
 Testing TJpg_Decoder library

Initialisation done.
....
For test only, removing file

Listing LittleFS files:
=================================================
  File name                              Size
=================================================
=================================================

Downloading /M81.jpg from https://i.imgur.com/C77RWcq.jpg
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200

[HTTP] connection closed or file end.
1743 ms to download

Listing LittleFS files:
=================================================
  File name                              Size
=================================================
  M81.jpg                                7857 bytes
=================================================

152 ms to draw to TFT

```

Bodmer, 2022-03-16:
The flicker on the screen does suggest power surge problems because the voltage level affects the LED brightness significantly. Glad to hear you have it working.