Bodmer/TFT_eSPI #1683: fatal error: FS.h: No such file or directory

spinalcode, 2022-03-05:
I get this error with every example I have tested on the rpi pico.

Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_RP2040.h:107:12: fatal error: FS.h: No such file or directory

Am I missing a specific library?

Bodmer, 2022-03-06:
FS.h is part of the Flash filing system. It is built into Earle Philhower's Arduino RP2040 board package and allows sketches to use LittleFS.

The error should not occur with the official Arduino mbed package due to the [#if def here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_RP2040.h#L104). Maybe you are using an older version of the TFT_eSPI libray or one those board packages?

Bodmer, 2022-03-06:
The current release has the if def missing so only works with Earle's board package. I will release a new version.

closed 2022-03-06 by Bodmer