Bodmer/TFT_eSPI #1685: (Touch Issue) RawZ always returns 4095

FerFemNemBem, 2022-03-06:
Hi,
I'm using an "ILI9341" (the second one now) and all of a suden the touch of them doesn't work anymore.
Calibration "magically" touches itself and the "Test_Touch_Controller.ino" reports:
...
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095
...
looks to me like fully touched (or even not connected at all?).

The strange thing - if I order and connetct a new display (second time until now) everything works at the beginning (rawZ reports reading between 10-20 untouched). All of a suddon after 2-3 days the rawZ readings are 4095.

My setup:
AZ-Delivery ESP32 Dev Kit C V4

"Setup1_ILI9341.h":
#define ILI9341_DRIVER
#define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

"User_Setup_Select.h":
#include <User_Setups/Setup1_ILI9341.h> 

"User_Setup.h":
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15
#define TFT_DC    2
#define TFT_RST   4
#define TOUCH_CS 17 (I have tried other ones like 16 already - same result)

Could it be, that these displays are all "crap" and died after some days ot is there anything that could be wrong from my side.
Could I do some more tests to figure out where these "4095"-readings are comming from?

Thank you!

BR,
FFNB

FerFemNemBem, 2022-03-07:
Hi,

I have to correct myself. The problem not occures "all of a sudden"!
I have connected an SD-Card-Reader at the same SPI-BUS later (of course with a different CS-Pin=. 
Now I've read here: https://github.com/Bodmer/TFT_eSPI/discussions/898 that this could cause troube.
Any chance to get the Touch controller working with an SD-Card-Reader on the same SPI-Bus connected? I couldn't find the solution inside the linked issue.
Is there any workaround? (On my display there is no R1 to bridge)

Best regards,
FFNB

Bodmer, 2022-03-07:
The test touch controller sketch will need to be modified if other SPI peripherals are on the bus. I am not sure what the root cause is. It is important to set the chip select pins high of all SPI devices at the start of setup() so that bus conflicts do not occur during the sequential initialisation. GPIO pins float to undefined levels when the pin has not been configured, many support boards do not prevent this by installing a pull up resistor. It could be that the touch controller needs a lower clock rate. Long wires can also cause problems, poor 0V connection as well.

Unfortunately it is difficult to advise in these situations.

Bodmer, 2022-03-07:
The linked issue is for particular ILI9488 and ST7796 boards, so I would not expect those conditions to apply in your case.

FerFemNemBem, 2022-03-07:
I have tried now:

- remove the (external) SD-Card-Reader (unsoldered SD-Card-CS-Pin 5) => no change
- unsoldered the MISO/SDO from the Display => no change
- checked all cable soldering (all cables are soldered fine - extra doublechecked GND)
- the cables are all about 15cm long and "good ones" - 26AWG silikon
- added pinMode "PULLUP" for all CS-Pins at the "setup()" beginning

Any idea what to doublecheck (until the next display arrives)?
Appended some pictures of the display:
[image]
[image]
[image]

Looks anything wrong on the Display backside itself? 
Any Ideas what I can check until the new display arrives? 

Thank you.
Best regards,
FFNB

Bodmer, 2022-03-07:
The 4095 falue is the 12 bit touch reading being all  's ie 0xFFF this typically indicates the touch controller is not driving the SPI bus. You could add a 1K pull down resistor in the touch controler output pin (typically called TDO) and then the value will be 0.

I have had many displays delivered from China and only had 2 bad one (cracked screens refunded for those) and one with a duff touch controller (suspect the ribbon under the display may be damaged but will likely damage the TFT if I try to lift it).  Personally I do not like the resistive touch screen as they are not very accurate and get scratched easily if a stylus is used. Having said that the library should work.

These displays are quite robust against anit-static damage too and take a fair bit of breadboarding abuse. They don't like 5V for long though .Most are fitted with a regulator 5V to 3.3V, that is U1 on your display. The connections and configuration (wrong display driver assumed) are where most folk go wrong even though they have checked a hundred times!  Sometimes it takes the use of test equipement like an oscilloscope figure out whats going on.

Ther are some rare versions of the library that maybe had a touch bug, but since you had it working that is unlikely to be the cause.

FerFemNemBem, 2022-03-07:
Strange thing: pulling down T_DO to GND does not change anything. The reading stays at 4095. The reading does never differ from 4095 even not by 1 or 2.
Now I will unsolder and bring the display back to a testing breadboard without other components connected.

Tomorrow the new one will arrive - I will report what happens with this one.

Thank you.

Best regards,
FFNB

closed 2022-03-08 by FerFemNemBem

garudaonekh, 2022-05-01:
Same issue, but removing MISO pin solve the problem.  
https://github.com/DustinWatts/FreeTouchDeck
Based on him, MISO is not required.

Bodmer, 2022-05-02:
@garudaonekh The problem display in the OP is an ILI9341 which are OK to connect MISO. The displays that are troublesome are documented here:
https://github.com/Bodmer/TFT_eSPI/discussions/898#discussion-1174065

Electronlibre2012, 2024-07-06:
> Same issue, but removing MISO pin solve the problem. https://github.com/DustinWatts/FreeTouchDeck Based on him, MISO is not required.

Thanks a lot @garudaonekh !

lost 4 hours to figure out why it doesnt work and read your POST! double check in the freetouchDeck : MISO pin is NOT connected to TFT, only with TDO of the Touchscreen!!!

thanks!