Bodmer/TFT_eSPI #1686: feature request: setSwapBytes from setup file

smarq8, 2022-03-06:
It would be nice to set '_swapBytes' from setup file.
I also noticed some functions are inconsistent about using swapBytes, some of them internally use "if(_swapBytes)" and other use swap passed by function parameter. It should be settled once and then forget about "should I use or not... Im not sure".

Bodmer, 2022-03-07:
The setSwapBytes function should only be needed for rendering 16 bit images which need their endianess changed. It is also needed to support processors that use different endianess for storage or SPI transfer. The default is byte swap false.

Bodmer, 2022-03-07:
I think it might cause confusion if byte swap is in the setup.

All examples should run without alteration on the supported processors. If that is not the case let me know.

closed 2022-03-07 by Bodmer