Bodmer/TFT_eSPI #1687: TFT_Button_Label_Datum Doesn't Compile

GeorgeIoak, 2022-03-07:
I got 2 errors when I tried to compile the reference example for a ESP32. The first was 

```
'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?
```
If I remove all the touch screen code and just have ot draw the buttons I get this error:

```
Arduino: 1.8.16 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Users\george\AppData\Local\Temp\arduino_modified_sketch_733088\TFT_Button_Label_Datum.ino: In function 'void drawButtons()':

C:\Users\george\AppData\Local\Temp\arduino_modified_sketch_733088\TFT_Button_Label_Datum.ino:62:35: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

                       KEY_TEXTSIZE);

                                   ^
```

I'm running the latest ESP32 core (2.02)

https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/TFT_Button_Label_Datum/TFT_Button_Label_Datum.ino

Bodmer, 2022-03-07:
The sketches that use screen touch functions require the touch controller chip select TOUCH_CS to be defined. Run Read_User_Setup and look at the serial output to check your setup.

GeorgeIoak, 2022-03-07:
On my 2nd test I modified the sketch to remove all the touch command so would the TOUCH_CS in Read_User_Setup still impact the compile in that case? That's the 2nd error listed above.

Bodmer, 2022-03-07:
The second message is a warning only - not an error, these tend to vary between processors, some report that warning others do not. The sketch wiil compile and run with warnings. Read_User_Setup will report if TOUCH_CS is defined.

closed 2022-03-07 by Bodmer