Bodmer/TFT_eSPI #1688: Problems implementing Custom FreeFont

Zelsof, 2022-03-07:
Hi Bodmer,

I am having troubles implementing a FreeFont to my project. Since I am completely out of ideas, I hope you can help me out.
I have a font, that I would like to use and I have it as an .h-file in the right size. I am working on an Arduino Mega 2560, using the ILI9488 display.

In the User_Setup.h I have:

```
#define ILI9488_DRIVER

#define TFT_CS   A0
#define TFT_DC   A6
#define TFT_RST  A5
#define TOUCH_CS A1

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts 

#define SPI_FREQUENCY  16000000
```

I use the subfolder `/fonts` (for Free_Fonts.h) and `/fonts/GFXFF` (for the fontfiles).

So my setup looks like:
```
#include "fonts/GFXFF/Open_24_Display_St_44pt.h"
#include "fonts/GFXFF/FreeMonoBold24pt_test.h"
#include "fonts/Free_Fonts.h"

#include <TFT_eSPI.h>
```

The "Free_Fonts.h" I edited by adding following line:
```
#define LCD_44px &Open_24_Display_St_reduced44pt7bBitmaps
#define Bold24 &FreeMonoBold24pt_testBitmaps

[...]#else
#define LCD_44px 1
#define Bold24 1
```

In the projekt I want to apply the font with:
```
tft.setTextColor(TFT_GREEN, TFT_BLACK);
tft.setFreeFont(Bold24);
tft.drawString("test", 100, 100, GFXFF);
```

So far I thought that I did everything right, but I am receiving an Error 
```
**FreeMonoBold24pt_test.h**
'GFXglyph' does not name a type
'GFXfont' does not name a type

**FreeMonoBold24pt_test.h**
'GFXglyph' does not name a type
'GFXfont' does not name a type
```

I also tried many different things, and even when there was no error, the screen displayed nothing. But when I use the Free_Font_Demo-Example, everything works fine.

So perhaps you see any mistake or maybe you can tell me if there is a "correct" way to implement own fonts. I would like to save them in my project folder, and not in TFT_espi, since everything is removed when updating.

Thank you in advance!

Bodmer, 2022-03-07:
Don't use the bitmaps array name, typically the array name to use is at the end of the font file. See example here:

https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Local_Custom_Fonts/MyFont.h#L3361

Zelsof, 2022-03-07:
Thanks,
I changed the arrayname. Unfortunately the errors are still the same:
```
**FreeMonoBold24pt_test.h**
'GFXglyph' does not name a type
'GFXfont' does not name a type

**FreeMonoBold24pt_test.h**
'GFXglyph' does not name a type
'GFXfont' does not name a type`
```

I tried to modifiy the order in the include-section to:
```
#include "fonts/Free_Fonts.h"

#include <TFT_eSPI.h>
#include "fonts/GFXFF/FreeMonoBold24pt_test.h"
```
When I do so, there are no errors, BUT from the point that the program passes the `tft.setFreeFont(Bold24);`, the program gets stuck and nothing happens anymore.

I especially used for the new font, that I am testing a copy but renamed version from your GFXFF-fonts, so I know that it should normally work. There is probably also no memory problem, since the compiler tells me I have about 164,486 bytes left.

Do you have any other idea, what can be the problem, or any test I can do?

Bodmer, 2022-03-07:
Start with the Local_Custom_Fonts example,  run it to check it works then just add your font file. Run again, copy the approach taken with the example, and make incremental changes and test each change.

Zelsof, 2022-03-07:
Wow, it worked out (by help of  Local_Custom_Fonts example)!
I deleted the `#include "fonts/Free_Fonts.h"` and included the font directly. So the file header is:
```
#include <TFT_eSPI.h>
#include "fonts/GFXFF/Open_24_Display_St_44pt.h"
#define GFXFF 1
#define TestFont &Open_24_Display_St_reduced44pt7b
```
With this it works this time:
```
tft.setFreeFont(TestFont);
tft.drawString(TempString, 220, (Display_Height - 1) / 2 - 20, GFXFF); 
tft.setFreeFont(NULL);
```

I don't really understand completely what went wrong, but with this solution I am fine.
Thank you very much!

Bodmer, 2022-03-07:
Gald it is working.

closed 2022-03-07 by Bodmer

Emcukcan, 2023-06-14:
I was having the same issue but it was having different reason to have error:

My code was like this and it was having GFX error as above
```
#include "orbitron.h" // same folder with sketch
#include <ArduinoJson.h>
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>
```
Simply I just changed code like this;

```
#include <ArduinoJson.h>
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>
#include "orbitron.h" // same folder with sketch
```
now it works this way.