Bodmer/TFT_eSPI #1690: Multiple SPI tasks issue

PierreDlrt, 2022-03-08:
Hi,

Thank you for your library, it is very complete and useful.
However, I am running into a problem when trying to make multiple freertos tasks communicate on the same SPI bus.
I have a simple program with 2 tasks : the first one writing periodically on the TFT screen, and the second one reading periodically eventual coordinates on the touch screen (code below).
Each task runs well individually but when running both a the same time, it gets stuck in the task dealing with the touch screen, more precisely at the first `spi.transfer()` attempt.

I have tried wrapping a mutex around every function using SPI and it is working but I can't do that for larger programs so it is not a permanent fix. And it looks like the Arduino SPI library is already managing SPI bus accesses so there should be no need to do it again.

I am using a Waveshare 4 inches Touch Shield with a ili9486 driver.

Am I using the library wrong ? or is it a bug ?

```
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void writeTFTtask(void* pvParam)
{
	xSemaphoreHandle mutexSPI = (xSemaphoreHandle) pvParam;
	uint16_t color = 0;

	for(;;)
	{
		// xSemaphoreTake(mutexSPI, portMAX_DELAY);
		tft.fillScreen(color);
		// xSemaphoreGive(mutexSPI);
		color = (color + 1)%0x10000;
		vTaskDelay(1/portTICK_PERIOD_MS);
	}
}

void readTouchtask(void* pvParam)
{
	xSemaphoreHandle mutexSPI = (xSemaphoreHandle) pvParam;
	uint16_t x, y;
	uint8_t isValidTouch;

	for(;;)
	{
		// xSemaphoreTake(mutexSPI, portMAX_DELAY);
		isValidTouch = tft.getTouch(&x, &y);
		// xSemaphoreGive(mutexSPI);
		if (isValidTouch) {
			Serial.printf("x = %d, y = %d\n", x, y);
		}
		vTaskDelay(1/portTICK_PERIOD_MS);
	}
}

//####################################################################################################
// Setup
//####################################################################################################
void setup() {
	Serial.begin(115200);

	// Set all chip selects high to avoid bus contention during initialisation of each peripheral
	digitalWrite(TOUCH_CS, HIGH); // Touch controller chip select (if used)
	digitalWrite(TFT_CS, HIGH); // TFT screen chip select

	tft.begin();
	
	tft.setRotation(3);
	tft.fillScreen(0);

	xSemaphoreHandle mutexSPI = xSemaphoreCreateBinary();
	xSemaphoreGive(mutexSPI);

	xTaskCreate(readTouchtask, "read touch pad", 4096, mutexSPI, 2, NULL);
	xTaskCreate(writeTFTtask, "write color", 4096, mutexSPI, 2, NULL);

}

//####################################################################################################
// Main loop
//####################################################################################################
void loop() {

}

```

Bodmer, 2022-03-09:
The library is not written to be thread safe!

If you keep touch and TFT access in the same task it should work but it is not something I have tried.

You may find using a third party touch library such as TFT_eTouch is a solution.

closed 2022-03-09 by Bodmer

Bodmer, 2022-03-09:
Out of interest I had a look at the [ESP32 website here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html).

In particular this warning:
[image]

This iindicates it is possible to make the SPI access code thread safe if each independant task is dedicated to one peripheral on the bus. It appears that this requres the use of SP bus handles but this handle designation is not exposed in Arduino calls. Thus the library would need to be modified to musch lower level calls. This is not something I will be trying in the short term but it is a possible future update project.