Bodmer/TFT_eSPI #1691: 3.5inch_RPi_LCD_(C) can not drivable

wangfreexx, 2022-03-09:
Although it is mentioned in the article, it can not be used in practice.

Bodmer, 2022-03-09:
Can you point out where this is mentioned.

The library supports some TFT displays designed for the Raspberry Pi (RPi) that are based on a ILI9486 or ST7796 driver chip with a 480 x 320 pixel screen. The ILI9486 RPi display must be of the Waveshare design and use a 16 bit serial interface based on the 74HC04, 74HC4040 and 2 x 74HC4094 logic chips. Note that due to design variations between these displays not all RPi displays will work with this library, so purchasing a RPi display of these types solely for use with this library is not recommended.

Bodmer, 2022-03-09:
I am guessing you have the single chip ILI9486 based125MHz display? Run Read_User_Setup and post the results.

wangfreexx, 2022-03-10:
Thank you for your reply, first of all, I am using the translation software for communication. I bought another version, can be found in this URL (http://www.lcdwiki.com/MHS-3.5inch_RPi_Display) , its driver is compatible with the version of the picture you send. His chip is the ILI9486, but there's no backlight control. I used the STM32 driver, but the screen just blinked. The selected file is (Setup33_RPi_ILI9486_STM32.h) .

And I don't understand the meaning of this sentence (“Run Read_User_Setup and post the results.”)

Bodmer, 2022-03-10:
Describe your hardware configuration (processor, board model etc), list all connections and pins used. A photo of your test setup will help too.
[Read_User_Setup](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) is a test and diagnostic example included in the library.

wangfreexx, 2022-03-14:
I use the device STM32F401CCU6, which is my setup：
[image]
I try to run Read_User_Setup,and here's what happened
[image]

wangfreexx, 2022-03-14:
[image]
I tried STM32F103C8T6, and again, I tried to slow down the SPI, but it was all white

Bodmer, 2022-03-14:
What are the power connections to the  TFT.    
List 5V, 3V3 and 0V connections.
The Read User Setup output looks OK, the pins for an STM32 board do not get reported correctly. Your setup looks good.

Bodmer, 2022-03-14:
I've checked the STM32 code and see that RPi type displays are currently not supported for the STM32 processors. RPi support has been included only for ESP32, ESP8266 and RP2040. I will look at adding support for STM32.

Bodmer, 2022-03-14:
The code for the RPi interface is present (so [ignore this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.h#L14)) and I have tested it in the past with one of the slower RPi based displays. I will setup a board and display to see if the library updates have caused problems.

Bodmer, 2022-03-14:
I have connected up an STM32F411 "Black Pill" board to an MHS-4.0 display. This is based on the ST7796 TFT controller as I do not have a MHS-3.2 ILI9486 based display.  This is the setup I have used:
```

#define STM32

#define RPI_DISPLAY_TYPE

//#define ILI9486_DRIVER
#define ST7796_DRIVER

#define TFT_SPI_PORT 1
#define TFT_MOSI PA7
#define TFT_MISO PA6 // Defined but NOT connected to RPi display
#define TFT_SCLK PA5
#define TFT_CS   PA1
#define TFT_DC   PA0
#define TFT_RST  PA2

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  20000000 
```

[image]

[image]

The setup is running fine. Can you check the connections in your setup and study the above settings to see if there is a difference somewhere.  It could be that the MHS-3.2 is somehow different in the interface details but I would expect them to be compatible.

Bodmer, 2022-03-15:
I think I may have found the problem. On some displays the hardware reset is not routed through to the display. The display does have a commanded software reset feature though. I have updated the master copy of the library, to test your setup then just make sure [these two lines are not commented out](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9486_Init.h#L11-L12).

wangfreexx, 2022-03-15:
[image]
As you can see, it really works. I changed the pin to remove the two-sentence comment, and we're good to go. Thank you very much.

Bodmer, 2022-03-15:
Great!

closed 2022-03-15 by Bodmer