Bodmer/TFT_eSPI #1694: Two 1 bit Sprites with different Colors

Fischkopppp, 2022-03-09:
Hi Bodmer, 
thank you for answering my previous question :)
I have two 1 bit Sprites and want to assign both different colors. One Sprite will be used a text window and the other Sprite simulates a scrollbar. However both Sprites will always have the same fg / bg color:

```
  balken.setColorDepth(1);
  balken.setBitmapColor(TFT_BLUE, TFT_DARKGREY);
  balken.createSprite(20,30);
  balken.fillSprite(TFT_DARKGREY);
  balken.fillRect(1,1,20,28,TFT_BLUE);

  speiseplan.setColorDepth(1);
  speiseplan.createSprite(219,h_sprite);
  speiseplan.setBitmapColor(TFT_BLACK, TFT_WHITE);
  speiseplan.fillSprite(TFT_WHITE);
  speiseplan.setTextColor(TFT_BLACK);
  speiseplan.setFreeFont(MYFONT32);
  speiseplan.setTextWrap(true,false);
  speiseplan.setCursor(0,0,GFXFF);
```

Bodmer, 2022-03-09:
Yes. The fg and bg colours are not unique to the Sprite instance.

You will need to set the colours immediately before pushing to the screen, as in [this example](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Sprite/Sprite_scroll_1bit/Sprite_scroll_1bit.ino#L85-L93).

I will update the library at some point to make the colours instance specific.

Bodmer, 2022-03-09:
It will take quite a few changes to the code to make the 1bpp colours sprite specific, so this is not a change I will be making for quite a while.

closed 2022-03-09 by Bodmer